zephyrproject-rtos/zephyr #15331: CONFIG_CODE_DATA_RELOCATION does not work with Cache enabled MCUs.

AntoineZen, 2019-04-10:
**Describe the bug**

When using the Code relocation (CONFIG_CODE_DATA_RELOCATION=y) with platforms that have a cache (tested on ARM Cortex-M7), leads to unreliable execution. When doing so, in the CPU crash in random exception (Hard-fault, MPU fault, Usage Fault, bus fault, etc..) with random cause (illegral access to EPSR...). The exception cause is never the same. A small change in the code, like changing a config or adding a variable or some code (even in un-executed area) can change the behavior or make the crash disappear. The bug is really random, so it can be hard to reproduce.

 By step-by-step debugging, it shows that the program execution does not match with the intended code when running from RAM. So eiter the machine code copied in RAM by `data_copy_xip_relocation()` is wrong, or the cache is doing something nasty.

If I comment-out the lines that enable the Cache (`SCB_EnableICache()` /` SCB_EnableDCache()`) the application run fine in all case. That seems to proof that the cache is involved. The issue was seen on the `mimxrt1050_evk` board, which has external serial flash, so disabling the cache is a bad idea with this board.

In the ARM Cortex-M product line, only M7 has a cache. Searching for `SCB_EnableICache()` in Zephyr's code base, It seems that only the iMX RT familly, SAM E70 and STM32F7 are using caches. All other board should not have this issue I think. 

I have try the following patch with no success:

```
diff --git a/kernel/init.c b/kernel/init.c
index 276912f649..67f8dda683 100644
--- a/kernel/init.c
+++ b/kernel/init.c
@@ -190,6 +190,11 @@ void z_data_copy(void)
        extern void data_copy_xip_relocation(void);
 
        data_copy_xip_relocation();
+#ifdef CONFIG_CPU_CORTEX_M7
+       /* Make sure everything is aligne between RAM & Cache */
+       SCB_CleanDCache();
+       SCB_InvalidateICache();
+#endif
 #endif /* CONFIG_CODE_DATA_RELOCATION */
 #ifdef CONFIG_USERSPACE
 #ifdef CONFIG_STACK_CANARIES
```

I'm happy if someone has an idea on how to fix this, I will have little time to work on this the next few weeks.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/application_development/code_relocation 
1. mkdir build; cd build
2. cmake -DBOARD=mximxrt1050_evk
3. make
4. make flash
4. If you are (un)lucky, you will have some fault. maybe not. If not try to change the code util it aprears.

**Expected behavior**
Not crashing

**Impact**
System halt. Having various/random error report in the console. 

**Screenshots or console output**
```
***** Booting Zephyr OS v1.14.0-rc1-1375-gbfafb75936cc *****
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x80001400
Faulting instruction address = 0xf4561b02
Fatal fault in essential thread! Spinning...
```
or 
```
***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x80000f44
Faulting instruction address = 0x4346
Fatal fault in thread 0x80000f44! Aborting.
```

**Environment (please complete the following information):**
 - OS: GNU/Linux Ubuntu 18.04
 - Toolchain e.g Zephyr SDK 0.10.0
 - Commit SHA  : 3a8db5549ab2a5d231900c6e727921aa852aa21e, but seen on other version.

**Additional context**
I was developing a Flash driver for the `mimxrt1050_evk` board at the time this first appeared. I will make a pull request with this code and link it to this issue.

wentongwu, 2019-04-12:
I tested SAM E70, and it works well.
Also it seems board mximxrt1050_evk isn't supported by zephyr so far.
yeah, this issue maybe highly relate to cache. but it highly depends on cache design and how cache is configured. 
So can you talk more about your soc's cache and how it's configured, also please attach you map file. Thanks

MaureenHelm, 2019-04-16:
I tried running this sample on mimxrt1050_evk and got the following result:
```
***** Booting Zephyr OS v1.14.0-rc3-176-g3a8db5549ab2 *****
Address of main function 0x600028f9

Address of function_in_sram2 0x81000005
Address of var_sram2_data 0x810001f4
Address of k_sem_give 0x81000001
Address of var_sram2_rodata 0x810001f0
Address of var_sram2_bss 0x810001f8

Address of function_in_sram 0x80000001
Hello World! mimxrt1050_evk
Address of custom_section, func placed using attributes 0x80001089
Address of custom_section data placed using attributes 0x80001084

Address of function_in_split_multiple 0x8100008d
Address of var_file3_sram_data 0x80000130
Address of var_file3_sram2_bss 0x810001fc
```
I used commit 3a8db5549ab2a5d231900c6e727921aa852aa21e and SDK 0.10.0 (same as you). The SHA in your boot banner doesn't seem to match this SHA.

wentongwu, 2019-04-29:
@AntoineZen from MaureenHelm's test, relocation code works well on board mimxrt1050_evk.

AntoineZen, 2019-04-29:
> @AntoineZen from MaureenHelm's test, relocation code works well on board mimxrt1050_evk.

As I said, the issue is random and thus hard to reproduce. Try to change anything in the code, and the issue might appear. Change again, the issue is gone. Maybe something related to memory alignement, cache... I was not able to spot it till now, thus this issue. If someone has more luck/expertise.

loicpoulain, 2019-05-04:
@AntoineZen do you reproduce the issue when relocating in ITCM?

rljordan, 2019-08-13:
Is this still an issue?