zephyrproject-rtos/zephyr #15334: Unable to reset nRF52840 from nRF9160 on pca10090 DK

lemrey, 2019-04-10:
It is desirable to be able to reset the nRF52840 on the pca10090 DK from the nRF9160, especially in situations where the two SoC communicate and their state needs to be synchronized (and given that only one reset button exists on the board).

Because the nRF52840 chip does not have a reset line connected to the nRF9160, a GPIO line can be used to issue a reset. Let nrf52840_pca10090 be reset via a dedicated line.

ioannisg, 2019-04-11:
@carlescufi are we addressing this in 1.14?

carlescufi, 2019-04-11:
@ioannisg yes, we are waiting for a PR from @lemrey

ioannisg, 2019-06-28:
@lemrey any updates?

lemrey, 2019-06-29:
This was done in: https://github.com/zephyrproject-rtos/zephyr/pull/15512

closed 2019-06-29 by lemrey