zephyrproject-rtos/zephyr #15343: tests/kernel/interrupt: Assertion Failure in test_prevent_interruption

cinlyooi-intel, 2019-04-11:
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=qemu_x86
3. make
4. make run

**Screenshots or console output**
```

***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-192-g100a2bba6c57 (delayed boot 1000ms) *****
Running test suite interrupt_feature
===================================================================
starting test - test_nested_isr
SKIP - test_nested_isr
===================================================================
starting test - test_prevent_interruption

 Assertion failed at zephyr.git/tests/kernel/interrupt/src/nested_irq.c:116: offload_function: (check_lock_new equal to check_lock_old)

Interrupt locking didn't work properly
FAIL - test_prevent_interruption
===================================================================
Test suite interrupt_feature failed.
===================================================================
RunID: ci-190410-1550-659:uytl
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain zephyr
 - Commit SHA or Version used: 110cab8aa37e0c08824cebae7047a042e56f8377

**Additional context**
Add any other context about the problem here.

overheat, 2019-04-11:
@cinlyooi-intel I can't reproduce it in my environment. It works for for me.
```
west build -b qemu_x86 tests/kernel/interrupt/
west build -t run
```
```
Booting from ROM..***** Booting Zephyr OS v1.14.0-rc3-193-g110cab8aa37e *****
Running test suite interrupt_feature
===================================================================
starting test - test_nested_isr
SKIP - test_nested_isr
===================================================================
starting test - test_prevent_interruption
PASS - test_prevent_interruption
===================================================================
Test suite interrupt_feature succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

cinlyooi-intel, 2019-04-11:
@overheat @wentongwu As it is triggered by an assertion, it definitely did happened. However, I think this might be one of the rarely occurred bug. This is the first time I see in about two weeks worth of daily triaging every possible failure in preparation for 1.14.

nashif, 2020-03-11:
cant reproduce.

closed 2020-03-11 by nashif