zephyrproject-rtos/zephyr #15348: ARM Cortex-M: SysTick: unhandled race condition

ioannisg, 2019-04-11:
It looks like we are not handling properly a possible race condition in ARM Cortex-M SysTick.

The implementation in `z_clock_set_timeout()` for TICKLESS mode performs the following steps:
1. Disables interrupts
2. Calculates SysTick re-load value based on the desired timeout
3. Re-configures and re-starts the SysTick timer to count-down towards this timeout.
4. Re-enables interrupts.

If, during steps (1) and (2) the SysTick actually wraps, an event will be triggered and the SysTick interrupt will be pending. As a result, as soon as we re-enable the interrupt, the ISR will be executed immediately; not after the desired timeout.

The event is rare but the impact is significant - we are advancing the clock immediately, when we should have been waited for the timeout.

Root-cause and solution:
Clearing the internal SysTick status flag (CTRL.COUNTFLAG) is not sufficient to clear a pending event; we also need to do it explicitly in ICSR (Clear the PENDSTSET bit).

Reproduce:
I have written a simple HW test case for ARM SysTick (a modified hello-world example) here:

https://github.com/ioannisg/zephyr/tree/drivers_systick_testcase/samples/hello_world

which demonstrates the case.

ioannisg, 2019-04-11:
FYI @andyross @pizi-nordic

ioannisg, 2019-04-11:
FYI @MaureenHelm @galak 
This affects tick-less operation in ARM platforms with SysTick as system timer.

pizi-nordic, 2019-04-11:
For me this issue is a next call for absolute time in kernel. As the timer is running all the time, the software will always be racing with it. So we either get more interrupts than we want or we lost/delay some of them.
And the code should be prepared to deal with such situations.

PR #15310 (2019-04-11): drivers: timer: systick: fix off-by-one setting
When the counter reaches zero, it reloads the value in
SYST_RVR on the next clock edge. This means that if the
LOAD value is N, the interrupt ("tick") is triggered
every N+1 cycles. Therefore, when we operate in non-
tickless mode, we need to configure the LOAD value
with CYC_PER_TICK - 1, in order to get an event
every CYC_PER_TICK cycles.

Fixes #15309 

Update: we address the race condition in TICKLESS mode in this pull-request, too.

Fixes #15348
closed 2019-04-11 by nashif