zephyrproject-rtos/zephyr #15349: stm32: uart fix pin pull-up

RandomReaper, 2019-04-11:
**Is your enhancement proposal related to a problem? Please describe.**
1. Pull-up are enable on the TX pins of stm32 uart, they are unnecessary and a waste of energy since TX pins are always driven by the peripheral.
2. Pull-ups are not enable on RX pins, this can be a problem when the uart is not connected (a shell uart for instance).
( see defines in drivers/pinmux/stm32/pinmux_stm32).

**Describe the solution you'd like**
Remove pull-ups on outputs and enable pull-ups on inputs.

**Describe alternatives you've considered**
Enable pull-ups only on dis-connectable uart rx pins.

overheat, 2019-04-11:
Good point!

erwango, 2021-11-04:
Closing, cf https://github.com/zephyrproject-rtos/hal_stm32/pull/88.
Re-open if required

closed 2021-11-04 by erwango