zephyrproject-rtos/zephyr #15361: nRF timer: investigate race condition when setting clock timeout in TICKLESS mode

ioannisg, 2019-04-11:
Relates to #15348 , please look there for more details.

The implementation of z_clock_set_timeout() in nRF system timer is as follows:

```
#ifdef CONFIG_TICKLESS_KERNEL
	ticks = (ticks == K_FOREVER) ? MAX_TICKS : ticks;
	ticks = MAX(MIN(ticks - 1, (s32_t)MAX_TICKS), 0);

	k_spinlock_key_t key = k_spin_lock(&lock);
	u32_t cyc, t = counter();

	/* Round up to next tick boundary */
	cyc = ticks * CYC_PER_TICK + counter_sub(t, last_count);
	cyc += (CYC_PER_TICK - 1);
	cyc = (cyc / CYC_PER_TICK) * CYC_PER_TICK;
	cyc += last_count;

	if (counter_sub(cyc, t) < MIN_DELAY) {
		cyc += CYC_PER_TICK;
	}

	set_comparator(cyc);
	k_spin_unlock(&lock, key);
#endif

```

The timer is running even after we lock interrupts. If an event is generated while we reprogram the comparator, we are going to get a timer IRQ immediately after we unlock interrupts, and this will make the clock slip.

T

ioannisg, 2019-04-11:
I have now investigated this in more detail and I think that the issue we have addressed in #15348 with respect to the SysTick timer does not appear in the nRF RTC timer. Here's is a detailed explanation:

Assume TICKLESS mode for the Kernel.
The problem we addressed in #15348 was, briefly, the fact that a timer wrap during z_clock_set_timeout() could make the timer give an interrupt immediately after exiting the z_clock_set_timeout(), which could erroneously "simulate" the case of a timeout expiring immediately.

This happened because of, two things:
1. The nature of SysTick:
     SysTick continues running, after the interrupt, emulating the exact same timeout duration
2. SysTick interrupt does not measure the current value and simply assumes that a full cycle has been completed since the last re-programming.

Neither of this is present in the nRF RTC:
a. After a Compare Event, the RTC keeps running in free mode, but it will only wrap in relatively long interval (512 seconds by default), so it is very hard to get an event interrupt while reprogramming the next timeout.
b. In the ISR, the RTC does not assume that a full cycle has passed, but performs a time sampling, anyways. So it won't simulate an immediate timeout expiration: it will actually calculate the elapsed time.

Therefore, I think that the nRF RTC system timer is robust.
I am reducing priority to low (this should not block the LTS release IMHO), but still invite @pizi-nordic to have a final say on this.

ioannisg, 2019-04-11:
@pizi-nordic please close this issue if you agree with my argumentation.

pizi-nordic, 2019-04-12:
Agreed.

closed 2019-04-12 by pizi-nordic