zephyrproject-rtos/zephyr #15363: Decoupling system timer from non-timer relaated areas

pizi-nordic, 2019-04-11:
At the moment the system timer frequency is used in many places due to lack of infrastructure needed to pass clock information to the drivers and SoC initialization code.

As we are going to have selectable system clock source, this becomes a a problem as system timer frequency might be no longer equal to system bus frequency driving given peripheral, leading to misconfigured devices.

For some peripherals (like UART and SPI), the solution is simple: There is a 'clock-frequency' property which can be used to pass required information.

Other peripherals are a bit problematic: For example the I2C already uses this property to specify I2C bus speed. So we have to introduce a new property to pass the information (see: https://elixir.bootlin.com/linux/latest/source/Documentation/devicetree/bindings/i2c/i2c-ocores.txt).

Also, we need a 'clock-frequency' property on CPU node in order to pass clocking information to SoC initialization code. Unfortunately it looks that our DTS infrastructure cannot extract information from CPU entries.

pizi-nordic, 2019-04-11:
FYI: @ioannisg, @carlescufi, @mbolivar

SebastianBoe, 2019-04-11:
> Unfortunately it looks that our DTS infrastructure cannot extract information from CPU entries.

Last I checked there was no DT binding for the CPU node so AFAICT all CPU DTS code is dead code.