zephyrproject-rtos/zephyr #15370: log_strdup() leaks memory if log message is filtered

andrewboie, 2019-04-11:
**Describe the bug**
If someone uses log_strdup() to safely log a string, and the logic in __LOG() decides that the message should be dropped, then the string is never freed and its memory in the log_strdup_pool is leaked.

**Expected behavior**
Any invocation of log_strdup() in an instrumentation point should eventually be freed even if the log message is filtered out at runtime.

**Impact**
Eventually, log_strdup() mem slabs get exhausted and we just log nulls.

andrewboie, 2019-04-11:
filed after discussion on Slack with @carlescufi and @jhedberg

carlescufi, 2019-04-11:
I think this is an issue regardless of whether `CONFIG_LOG_RUNTIME_FILTERING` is enabled or not, since the arguments are still evaluated thanks to `log_printf_arg_checker`

nordic-krch, 2019-04-12:
That sounded scary but I don't think that such thing will happen. That is because __LOG macro finally resolves to something like:
```
if (flag) {
    log_func(..., arg0, log_strdup(arg1), ...);
} else if (false) {
   log_printf_arg_checker(arg0,arg1,...);
}
```
so even in runtime if flag is resolved to false arguments will not be evaluated. Same goes for prinf_arg_checker which is never executed (only goes through compiler).

That was the intention to ensure that arguments are not evaluated when not used.

To be on the same side i even tried this, added log message to `log_strdup` function and used in  the example. `log_strdup()` was not called when log was disabled for given module.

carlescufi, 2019-04-12:
@nordic-krch that makes sense. I will close this issue, thanks for the explanation

closed 2019-04-12 by carlescufi