zephyrproject-rtos/zephyr #15374: PR 15230 introduces test failure for particle_argon

pabigot, 2019-04-11:
git bisect identifies 7b1ee5cf13dc4f130b556534bd4c8dd78ceec6c6 as introducing this failure, which blocks testing of #15369.
```
lilith[402]$ git reset --hard 7b1ee5cf13dc4f130b556534bd4c8dd78ceec6c6
HEAD is now at 7b1ee5cf13 tests: CONFIG_TEST_USERSPACE now off by default
lilith[403]$ rm -rf argon ; west build -d argon -b particle_argon tests/drivers/spi/spi_loopback/
source directory: /mnt/devel/external/zp/zephyr/tests/drivers/spi/spi_loopback
build directory: /mnt/devel/external/zp/zephyr/argon (created)
BOARD: particle_argon
Zephyr version: 1.14.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Selected BOARD particle_argon
-- Found west: /home/pab/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
-- Loading /mnt/devel/external/zp/zephyr/boards/arm/particle_argon/particle_argon.dts as base
-- Overlaying /mnt/devel/external/zp/zephyr/dts/common/common.dts
particle_argon.dts.pre.tmp:312.23-315.5: Warning (simple_bus_reg): /soc/virtualcom: missing or empty reg/ranges property
Parsing Kconfig tree in /mnt/devel/external/zp/zephyr/tests/drivers/spi/spi_loopback/Kconfig
Loading /mnt/devel/external/zp/zephyr/boards/arm/particle_argon/particle_argon_defconfig as base
Merging /mnt/devel/external/zp/zephyr/tests/drivers/spi/spi_loopback/prj.conf
Configuration written to '/mnt/devel/external/zp/zephyr/argon/zephyr/.config'
-- Cache files will be written to: /home/pab/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
Including module: tinycbor in path: /mnt/devel/external/zp/modules/lib/tinycbor
CMake Warning at ../../../../CMakeLists.txt:1468 (message):
  

        ------------------------------------------------------------
        --- WARNING:  __ASSERT() statements are globally ENABLED ---
        --- The kernel will run more slowly and use more memory  ---
        ------------------------------------------------------------

-- Configuring done
CMake Error at ../../../../cmake/extensions.cmake:357 (add_library):
  No SOURCES given to target: drivers__spi
Call Stack (most recent call first):
  ../../../../cmake/extensions.cmake:334 (zephyr_library_named)
  ../../../../drivers/spi/CMakeLists.txt:1 (zephyr_library)

-- Build files have been written to: /mnt/devel/external/zp/zephyr/argon
ERROR: command exited with status 1: /usr/local/bin/cmake -B/mnt/devel/external/zp/zephyr/argon -H/mnt/devel/external/zp/zephyr/tests/drivers/spi/spi_loopback -GNinja -DBOARD=particle_argon
run as "west -v build -d argon -b particle_argon tests/drivers/spi/spi_loopback/" for a stack trace
```

nashif, 2019-04-12:
@andrewboie any idea what is going on here? This is a weird one.

mike-scott, 2019-04-12:
Looks like SPI is enabled but no HW driver is selected.

andrewboie, 2019-04-12:
Userspace is off so the syscall handlers aren't built, so there's nothing to build at all for the driver. @SebastianBoe any idea?

nashif, 2019-04-12:
@mike-scott 

the actual offending commit is dd70c6bb92f5da6c8af59665785e70f71647cfc8

you should have 

```
if SPI

config SPI_2
	default y

endif
```

in Kconfig.defconfig

galak, 2019-04-16:
This issue might be related: #15495

PR #15496 (2019-04-16): boards: arm: particle_*: default-enable SPI_2 when SPI enabled
SPI is not normally enabled, but some tests assume that there's a device
available.  Conditionally enable SPI_2 which is associated with the
on-board flash.

Closes #15374
commit db365c43 (2019-04-16): boards: arm: particle_*: default-enable SPI_2 when SPI enabled

SPI is not normally enabled, but some tests assume that there's a device
available.  Conditionally enable SPI_2 which is associated with the
on-board flash.

Closes #15374
closed 2019-04-16 by galak