zephyrproject-rtos/zephyr #15379: foundries.io CI: sanitycheck fail on nrf52

mike-scott, 2019-04-12:
**Describe the bug**
Due to PR: https://github.com/zephyrproject-rtos/zephyr/commit/3e255e968a250b0099518cc2edb539183dd1a98f
sanitycheck now fails on nrf52_blenano2 (log below in )

Related CI failure:
https://ci.foundries.io/projects/zephyr/builds/5694/sanity-test-nrf52/

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/kernel/mem_protect/stackprot
2. mkdir build; cd build
3. cmake -DBOARD=nrf52\_blenano2
4. make flash
5. See above error

**Expected behavior**
Pass sanitycheck like previous to this commit

**Impact**
It's a regression for CI testing.. all of nRF52 will probably fail this test as a result

**Screenshots or console output**
```
 94/153 nrf52_blenano2            tests/kernel/mem_protect/stackprot/kernel.memory_protection FAILED: timeout
../tmp/outdir/nrf52_blenano2/tests/kernel/mem_protect/stackprot/kernel.memory_protection/handler.log
�***** Booting Zephyr OS v1.14.0-rc3-230-g1897896e9739 *****
Starts alternate_thread
alternate_thread: Input string is too long and stack overflowed!

***** Stack Check Fail! *****
Current thread ID = 0x20000068
Faulting instruction address = 0x8a54
Fatal fault in thread 0x20000068! Aborting.
Running test suite stackprot
===================================================================
starting test - test_stackprot
***** MPU FAULT *****
  Stacking error (context area might be not valid)
  Data Access Violation
  MMFAR Address: 0x20001c10
***** Stack Check Fail! *****
Current thread ID = 0x20000344
Faulting instruction address = 0x97e0
Fatal fault in essential thread! Spinning...
```

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK 0.10.0
 - Commit SHA: today's master (a942fcc23ed848fa6bcd16f5b45c996d7699f7b0)

aescolar, 2019-04-12:
@mike-scott : From your description this seems to not be related to sanitycheck, but to this particular test? Also, Is this test is happening in the Zephyr master branch? If so, the problem would not be related to your CI, or?
If that is the case, could you change the title to be a bit more clear that it just relates to this test and particular board?
(Something like  "tests/kernel/mem_protect/stackprot fails on nrf52_blenano2" )

mike-scott, 2019-04-12:
@aescolar sanitycheck runs this for nrf52 hardware.  The "easy" way to reproduce is to run just this test.  The actual regression was found via sanitycheck for real HW.

I don't see the issue with the description.

mike-scott, 2019-04-12:
Note: this test will probably fail on a lot more hardware than just nrf52_blenano2.

I'll adjust the title to mention the test specifically.

wentongwu, 2019-04-12:
@mike-scott how you trigger the CI test, Thanks? Actually this change passed the CI test before checkin.

nashif, 2019-04-12:
> @mike-scott Actually this change passed the CI test before check in. Anyway I will fix it ASAP.

CI does not run on real HW

PR #15417 (2019-04-12): tests/kernel/mem_protect/stackprot: stack size adjust
revert commit 3e255e968 which is to adjust stack size
on qemu_x86 platform for coverage test, but break other
platform's CI test.

Fixes: #15379.
commit 8646a8e4 (2019-04-12): tests/kernel/mem_protect/stackprot: stack size adjust

revert commit 3e255e968 which is to adjust stack size
on qemu_x86 platform for coverage test, but break other
platform's CI test.

Fixes: #15379.
closed 2019-04-12 by nashif