zephyrproject-rtos/zephyr #15388: [Coverity CID :197616]Null pointer dereferences in /subsys/testsuite/ztest/src/ztest_mock.c

aasthagr, 2019-04-12:
Static code scan issues seen in File: /subsys/testsuite/ztest/src/ztest_mock.c
 Category: Null pointer dereferences
 Function: z_ztest_get_return_value
 Component: Other
 CID: 197616
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-04-12:
*** CID 197616:  Null pointer dereferences  (FORWARD_NULL)
/subsys/testsuite/ztest/src/ztest_mock.c: 226 in z_ztest_get_return_value()
220     
221     if (!param) {
222     PRINT("Failed to find return value for function %s\n", fn);
223     ztest_test_fail();
224     }
225     
>>>     CID 197616:  Null pointer dereferences  (FORWARD_NULL)
>>>     Dereferencing null pointer "param".
226     value = param->value;
227     free_parameter(param);
228     
229     return value;
230     }
231

aescolar, 2019-04-12:
False positive. Confirmed by @mtpr-ot ("the ztest_test_fail() will abort the process if param == NULL, so no NULL pointer dereference is possible."). Flagged as false positive in the tool..

closed 2019-04-12 by aescolar