zephyrproject-rtos/zephyr #15391: [Coverity CID :197613]Possible Control flow issues in /subsys/net/ip/net_core.c

aasthagr, 2019-04-12:
Static code scan issues seen in File: /subsys/net/ip/net_core.c
 Category: Possible Control flow issues
 Function: services_init
 Component: Networking
 CID: 197613
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-04-12:
*** CID 197613:  Possible Control flow issues  (DEADCODE)
/subsys/net/ip/net_core.c: 436 in services_init()
430     static inline int services_init(void)
431     {
432     int status;
433     
434     status = net_dhcpv4_init();
435     if (status) {
>>>     CID 197613:  Possible Control flow issues  (DEADCODE)
>>>     Execution cannot reach this statement: "return status;".
436     return status;
437     }
438     
439     dns_init_resolver();
440     
441     net_shell_init();

tbursztyka, 2019-04-17:
@jukkar can you mark this as a non-issue in coverity? When dhcpv4 is not enabled, indeed that code flow cannot happen (but gcc is smart enough to optimize that path).

jukkar, 2019-04-17:
Agreed, this is a non-issue. I will mark it.

closed 2019-04-17 by jukkar