zephyrproject-rtos/zephyr #15392: [Coverity CID :197612]Integer handling issues in /lib/os/printk.c

aasthagr, 2019-04-12:
Static code scan issues seen in File: /lib/os/printk.c
 Category: Integer handling issues
 Function: z_vprintk
 Component: Other
 CID: 197612
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-04-12:
*** CID 197612:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
/lib/os/printk.c: 197 in z_vprintk()
191     u32_t u;
192     
193     if (long_ctr == 0) {
194     u = va_arg(ap, unsigned int);
195     } else if (long_ctr == 1) {
196     long lu = va_arg(ap, unsigned long);
>>>     CID 197612:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
>>>     "lu > 2147483647" is always false regardless of the values of its operands. This occurs as the logical operand of "if".
197     if (lu > INT32_MAX) {
198     print_err(out, ctx);
199     break;
200     }
201     u = (u32_t)lu;
202     } else {

galak, 2019-04-12:
This is a duplicate of 13843, Coverity triggered due to:

commit 4aa4883 (upstream/topic-sensors)
Author: Patrik Flykt patrik.flykt@intel.com
Date: Thu Mar 14 19:42:51 2019 -0600

subsystems: Rename reserved function names

closed 2019-04-12 by galak