zephyrproject-rtos/zephyr #15393: [Coverity CID :197611]Integer handling issues in /lib/os/printk.c

aasthagr, 2019-04-12:
Static code scan issues seen in File: /lib/os/printk.c
 Category: Integer handling issues
 Function: z_vprintk
 Component: Other
 CID: 197611
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-04-12:
*** CID 197611:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
/lib/os/printk.c: 166 in z_vprintk()
160     s32_t d;
161     
162     if (long_ctr == 0) {
163     d = va_arg(ap, int);
164     } else if (long_ctr == 1) {
165     long ld = va_arg(ap, long);
>>>     CID 197611:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
>>>     "ld > 2147483647" is always false regardless of the values of its operands. This occurs as the logical first operand of "||".
166     if (ld > INT32_MAX || ld < INT32_MIN) {
167     print_err(out, ctx);
168     break;
169     }
170     d = (s32_t)ld;
171     } else {

galak, 2019-04-12:
This is a duplicate of 13843, Coverity triggered due to:

commit 4aa48833d812ca56b62226462d48da24fb9bf756 (upstream/topic-sensors)
Author: Patrik Flykt <patrik.flykt@intel.com>
Date:   Thu Mar 14 19:42:51 2019 -0600

    subsystems: Rename reserved function names

closed 2019-04-12 by galak