zephyrproject-rtos/zephyr #15400: [Coverity CID :197604]Memory - corruptions in /tests/kernel/mem_protect/userspace/src/main.c

aasthagr, 2019-04-12:
Static code scan issues seen in File: /tests/kernel/mem_protect/userspace/src/main.c
 Category: Memory - corruptions
 Function: stack_buffer_scenarios
 Component: Tests
 CID: 197604
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-04-12:
*** CID 197604:  Memory - corruptions  (ARRAY_VS_SINGLETON)
/tests/kernel/mem_protect/userspace/src/main.c: 991 in stack_buffer_scenarios()
985      * the stack pointer.
986      *
987      * First test does direct read & write starting at the estimated
988      * stack pointer up to the highest addresses in the buffer
989      */
990     stack_ptr = &val;
>>>     CID 197604:  Memory - corruptions  (ARRAY_VS_SINGLETON)
>>>     Using "pos" as an array.  This might corrupt or misinterpret adjacent memory locations.
991     for (pos = stack_ptr; pos < stack_end; pos++) {
992     /* pos is volatile so this doesn't get optimized out */
993     val = *pos;
994     *pos = val;
995     }
996

andrewboie, 2019-04-12:
this is a false positive, I dismissed in Coverity

closed 2019-04-12 by andrewboie