zephyrproject-rtos/zephyr #15411: tests/kernel/critical:  Continuous reboot at test_critical

cinlyooi-intel, 2019-04-12:
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=sam_e70_xplained
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-231-ga69cb036c4e4 (delayed boot 1000ms) *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-231-ga69cb036c4e4 (delayed boot 1000ms) *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc3-231-ga69cb036c4e4 (delayed boot 1000ms) *****
Running test suite kernel_critical
===================================================================
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: a69cb036c4e457e66e9bd94549b8151e74229598

ioannisg, 2019-04-12:
Some issue with the watchdog, perhaps?

nashif, 2019-04-12:
bisected and found f0835674a31da824c3b9cfb6a68376960975e07d to be guilty from pr #14961

MaureenHelm, 2019-04-12:
Reproduced on frdm_k64f

nashif, 2019-04-12:
that bisect is wrong , tested on v1.14.0-rc3-41-gc240b69956 and it worked fine, looking into it

MaureenHelm, 2019-04-12:
@ioannisg It looks like a systick occurring during mpu dynamic reconfiguration

```
Program received signal SIGSEGV, Segmentation fault.
__usage_fault () at /home/maureen/zephyrproject/zephyr/arch/arm/core/fault_s.S:97
97		eors.n r0, r0
(gdb) bt
#0  __usage_fault () at /home/maureen/zephyrproject/zephyr/arch/arm/core/fault_s.S:97
#1  <signal handler called>
#2  console_out (c=42) at /home/maureen/zephyrproject/zephyr/drivers/console/uart_console.c:110
#3  0x00004aa8 in z_vprintk (out=out@entry=0xb75 <char_out>, 
    ctx=ctx@entry=0x20001e4c <_interrupt_stack+1740>, 
    fmt=fmt@entry=0x5d05 "***** BUS FAULT *****\n", ap=...)
    at /home/maureen/zephyrproject/zephyr/lib/os/printk.c:113
#4  0x00000c42 in vprintk (fmt=fmt@entry=0x5d05 "***** BUS FAULT *****\n", ap=..., 
    ap@entry=...) at /home/maureen/zephyrproject/zephyr/lib/os/printk.c:335
#5  0x00004d2a in printk (fmt=0x5d05 "***** BUS FAULT *****\n")
    at /home/maureen/zephyrproject/zephyr/lib/os/printk.c:385
#6  0x00001958 in BusFault (esf=esf@entry=0x20001eb8 <_interrupt_stack+1848>, fromHardFault=0)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/fault.c:343
#7  0x00001c92 in FaultHandle (fault=<optimized out>, esf=0x20001eb8 <_interrupt_stack+1848>)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/fault.c:628
#8  _Fault (esf=0x20001eb8 <_interrupt_stack+1848>, exc_return=<optimized out>)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/fault.c:862
#9  0x00001ebe in __usage_fault ()
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/fault_s.S:147
#10 <signal handler called>
#11 0x0000162a in z_clock_isr (arg=0x4 <z_finalize_fd+4>)
    at /home/maureen/zephyrproject/zephyr/drivers/timer/cortex_m_systick.c:91
#12 <signal handler called>
#13 0x00002168 in region_init (index=index@entry=4, 
    region_conf=region_conf@entry=0x20001f38 <_interrupt_stack+1976>)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:104
#14 0x00002224 in mpu_sram_partitioning (p_region=0x20001f84 <_interrupt_stack+2052>, 
    index=<optimized out>)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:225
#15 mpu_configure_regions (regions=regions@entry=0x20001f80 <_interrupt_stack+2048>, 
    regions_num=regions_num@entry=1 '\001', start_reg_index=<optimized out>, 
    do_sanity_check=do_sanity_check@entry=false)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:264
#16 0x00002386 in mpu_configure_dynamic_mpu_regions (regions_num=1 '\001', 
    dynamic_regions=0x20001f80 <_interrupt_stack+2048>)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:344
#17 arm_core_mpu_configure_dynamic_mpu_regions (
    dynamic_regions=dynamic_regions@entry=0x20001f80 <_interrupt_stack+2048>, 
    regions_num=regions_num@entry=1 '\001')
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/cortex_m/mpu/nxp_mpu.c:554
#18 0x000020dc in z_arch_configure_dynamic_mpu_regions (thread=<optimized out>)
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/cortex_m/mpu/arm_core_mpu.c:244
#19 0x00001908 in __pendsv ()
    at /home/maureen/zephyrproject/zephyr/arch/arm/core/swap_helper.S:191
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

nashif, 2019-04-12:
2895da02a44980e901fbcdd7ec87edc2016605c5

nashif, 2019-04-12:
@MaureenHelm @ioannisg  got to the same systick commit.

ioannisg, 2019-04-12:
OK, I am going to look at it now. Worst case we will have to revert that SysTick event-clear commit

ioannisg, 2019-04-12:
@MaureenHelm thanks for the GDB output. Could you also post the test print log?

A bit confused: SystTick interrupt during PendSV (Context-switch) should be happening all the time.

ioannisg, 2019-04-12:
Now I am totally confused. The test is passing on nRF52, nRF52840 on current master.
It passes when I change the nRF52 to use the SysTick instead of the nRF RTC timer.
At the moment I have no access to non-Nordic HW to test this one.

ioannisg, 2019-04-12:
```
***** Booting Zephyr OS v1.14.0-rc3-240-g7f478a5d29f1 *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
PASS - test_critical
===================================================================
Test suite kernel_critical succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

andrewboie, 2019-04-12:
I tested on two boards, sam_e70_xplained and frdm_k64f. It reboots over and over on both.

@ioannisg do you have any frdm_k64f boards?

ioannisg, 2019-04-12:
At my current location, no, @andrewboie .

MaureenHelm, 2019-04-12:
The test hangs on mimxrt1050_evk.

If I revert commit 2895da02a44980e901fbcdd7ec87edc2016605c5, the test passes on frdm_k64f and mimxrt1050_evk.

wentongwu, 2019-04-12:
I tested it yesterday with sam_e70_xplained also, it behaved as Cinly mentioned.

nashif, 2019-04-16:
posted a workaround and kept bug open.

cinlyooi-intel, 2019-04-23:
Confirmed working on 8e307a3ad9648035dcf84f05466c5bd9f6166c7a. Kept bug open as per @nashif  comment above (https://github.com/zephyrproject-rtos/zephyr/issues/15411#issuecomment-483469642)

ioannisg, 2019-06-28:
I want to address this one @nashif . Could we just call wdt_feed() inside the while/for loop of this test?

ioannisg, 2019-06-28:
Alternatively, disable the watchdog for this test...

PR #17373 (2019-07-11): tests: kernel: critical: alternative test-case for SAM SoCs with WDT
In SAM SoCs Watchdog is selected by default and runs
with some default configuration, unless the build sets
CONFIG_WDT_DISABLE_AT_BOOT. As the tests/kernel/critical
takes relatively large amount of time to complete, the
watchdog (that is never fed in the test) will eventually
trigger a reset. As a result the test keeps restarting
continuously and never completes. We want to run the
test on SAM SoCs, so we do the following:
- filter our the SAM SoCs with the SAM WDT from the
  default build
- introduce an alternative test-case [... cut]
closed 2019-07-11 by MaureenHelm