zephyrproject-rtos/zephyr #15419: reset and halt function

StefJar, 2019-04-12:
It would be great to have an "reset" or "halt" function, which can be called form user space.

Use cases:
-  something got bad inside the app
- own firmware upgrades via own bootloader written with ZephyrOS

aescolar, 2019-04-12:
`sys_reboot` does not fulfill your needs?

StefJar, 2019-04-12:
thanks - almost there
1st) for me a bit confusing, that a "sys_reboot" function can be activated under boot options
2nd) a halt function which stops the system but does not a reboot. This is useful for error search

aescolar, 2019-04-12:
@StefJar I assign @andrewboie as I hope he can help be more useful in this discussion

andrewboie, 2019-04-12:
`> a halt function which stops the system but does not a reboot. This is useful for error search

k_panic() does this

carlescufi, 2019-04-12:
Summarizing the answers:

- `sys_reboot()` will reboot the MCU
- `k_panic()` will halt the system

closed 2019-04-12 by carlescufi

StefJar, 2019-04-16:
maybe we can add this information to the kernel API documentation