zephyrproject-rtos/zephyr #15421: setting the system/logging time

StefJar, 2019-04-12:
So far there is no access to set the system/log time via function call. This can be extremely useful, when having distributed systems. Fusing the logs from the different nodes is done via merge and sort by time. To get a full log - let me call it "system log"  - of the whole system(=multiple devices) a common reference time is needed.

nordic-krch, 2019-04-15:
do you mean providing external function for timestamping?

https://github.com/zephyrproject-rtos/zephyr/blob/06ab394f05c0d7561a815cbaf4e7131fe91510f1/include/logging/log_ctrl.h#L65

StefJar, 2019-04-16:
Don't get me wrong - I like to have 2 additional function at the API:
- set_systime(date d, time t)
- get_timestamp(timestamp * t)

**Why I like to have set_systime?**
my typical log:
```
***** Booting Zephyr OS v1.14.0-rc3-258-g5307858e4e83 *****

[00:00:00.134,000] <inf> MAIN: start

[00:00:00.136,000] <dbg> MAIN.main: init leds

[00:00:00.138,000] <inf> SINK: init data sink

[00:00:00.140,000] <dbg> MAIN.main: init buttons

[00:00:00.143,000] <dbg> MAIN.main: init usb

[00:00:00.145,000] <inf> USB_ACM: init USB ACM

[00:00:00.148,000] <inf> USB_ACM: Baudrate detected: 115200

[00:00:00.151,000] <dbg> USB_ACM.usbACM_init: try to set baudrate to 115200

```
if my system(multiple devices) have a central time server that syncs every time in my system. Imagine something like [NTP](https://en.wikipedia.org/wiki/Network_Time_Protocol)

For example if you building a embedded Linux and activate NTP the linux system will sync its time when it connects via Ethernet/WLAN.

**Why I like to have get_timestamp?**
When sampling and processing is not taking place just in time(because we maybe like to store the data and transmit it), we need to add a timestamp for that. Whit this timestamp we can check if we lost data  or get the sample frequency.
Usually the timestamp has a resolution of 1ns-1ms. [Linux](http://man7.org/linux/man-pages/man2/clock_gettime.2.html) & [Windows](https://docs.microsoft.com/en-us/windows/desktop/sysinfo/acquiring-high-resolution-time-stamps) are providing different APIs for high resulution times.

These problems are not fully related to logging. It is more a general question, how to provide a consistent API for the user regarding date/time functions. Logging and sampling are my current fields of interest but this also goes to event detection(button pressed for 100ms)  etc.

StefJar, 2019-04-25:
I added a sample implementation for the stm32f4 arch
https://github.com/StefJar/zephyr_stm32_rtc_driver

carlescufi, 2020-03-17:
@StefJar can you please send a Pull Request for this?