zephyrproject-rtos/zephyr #15429: shields: improve cmake to define/extract pinmux and defconfig info

avisconti, 2019-04-12:
## Introduction

Currently shields are able to provide dts specific info thru an overlay file, which gets parsed
by cmake in order to extend the host board dts with the shield information. 

---
Ex:
boards/shields/x_nucleo_iks01a1/x_nucleo_iks01a1.overlay

/*
 *
 * Copyright (c) 2018 Linaro Limited
 *
 * SPDX-License-Identifier: Apache-2.0
 */

&arduino_i2c {

	hts221@5f {
		compatible = "st,hts221";
		reg = <0x5f>;
		label = "HTS221";
	};

	lps25hb-press@5d {
		compatible = "st,lps25hb-press";
		reg = <0x5d>;
		label = "LPS25HB";
	};

	lis3mdl-magn@1e {
		compatible = "st,lis3mdl-magn";
		reg = <0x1e>;
		label = "LIS3MDL";
	};

	lsm6ds0@6b {
		compatible = "st,lsm6ds0";
		reg = <0x6b>;
		label = "LSM6DS0";
	};
};

---

The current implementation does not provide similar overlay feature for pinmux and defconfig
definitions, which requires a direct change in the board files directly.

The proposal is to extend cmake capabilities in order to be able to specify in a standard
(arduino-like) way the pinmux as well as the Kconfig.defconfig extensions.

For example in order to support the x_nucleo_cca02m1 microphone board and test it
on Nucleo F411RE, I had to extend

boards/arm/nucleo_f411re/Kconfig.defconfig:

if I2S

config I2S_STM32
       default y

config I2S_STM32_USE_PLLI2S_ENABLE
       default y

config I2S_STM32_PLLI2S_PLLM
       default 8

config I2S_STM32_PLLI2S_PLLN
       default 192

config I2S_STM32_PLLI2S_PLLR
       default 3

config I2S_1
       default y

config USE_STM32_LL_TIM
       default y

endif # I2S

if DMA

config DMA_STM32F4X
       default y

endif # DMA

Similarly the F411RE pinmux definition.

avisconti, 2019-04-12:
@erwango @galak 
Can you take a look to this RFC, pls?
I'm not sure how to proceed here.
I'm about to support a ST shield with microphones on-board.

mike-scott, 2019-04-12:
@avisconti: can you take a look here? https://github.com/zephyrproject-rtos/zephyr/pull/14057

avisconti, 2019-04-12:
> @avisconti: can you take a look here? #14057

It looks like exactly what I was talking about here. With much more details of what I was able to
describe here.

So, the work seems a bit spread into different needa, but I'm particularly looking at the "boards: shields: introduce SparkFun u-blox SARA-R4 shield" commit, where I can see that you are addressing some of my points (like how to extend the host board config file)

avisconti, 2019-04-12:
@mike-scott 
Correct me if I'm wrong. It seems that in the shield you may define 

1. a per-board dts overlay (.overlay) file and/or a config (.conf) file
2. a shield dts overlay file and/or config file common for all boards

Next week I'll take a look to it more deeply. Thx for pointing me to your work!

avisconti, 2019-06-26:
Addressed by #14057

erwango, 2019-11-12:
@avisconti, should we keep this open?

gmarull, 2022-08-11:
pinmux is about to be deprecated and most platforms have moved to pinctrl, so closing. please re-open with new details if applicable.

closed 2022-08-11 by gmarull