zephyrproject-rtos/zephyr #15435: device fails to boot when spi max frequency set above 1000000

arewhyan, 2019-04-12:
I'm using MCUboot on  a device w/ an STM32L4 and an external W25QXX flash part connected via SPI.

        spi-flash@0 {
                #address-cells = <1>;
                #size-cells = <1>;
                compatible = "winbond,w25q16";
                reg = <0>;
                spi-max-frequency = <1000000>;
                label = "W25QXXDV";

With the spi-max-frequency set to 1000000, the device boots as expected. I added some log messages to capture the selected clock configuration.

[00:00:00.000,000] [1;31m<err> spi_ll_stm32: clock freq 80000000Hz
[00:00:00.001,000] [1;31m<err> spi_ll_stm32: frequency 1000000Hz, max 40000000Hz, min 312500Hz
[00:00:00.010,000] [1;31m<err> spi_ll_stm32: Installed config 0x20003798: freq 625000Hz (div = 128), mode 0/0/0, slave 0
***** Booting Zephyr OS 1.14.0-rc1 *****
[00:00:00.024,000] [0m<inf> mcuboot: Starting bootloader.

If I set spi-max-frequency to anything larger than 1000000, the device fails to boot. I'm not able to debug it at the moment, because the debugger also crashes.

[00:00:00.000,000] [1;31m<err> spi_ll_stm32: clock freq 80000000Hz
[00:00:00.001,000] [1;31m<err> spi_ll_stm32: frequency 2000000Hz, max 40000000Hz, min 312500Hz
[00:00:00.010,000] <ESC>[1;31m<err> spi_ll_stm32: Installed config 0x20003798: freq 1250000Hz (div = 64), mode 0/0/0, slave 0

This is where the device hangs, after selecting a frequency of 1250000 MHz. Couldn't find any other similar issues or documented SPI frequency limitations.

Following is the clock configuration I have selected.

# clock configuration
CONFIG_CLOCK_CONTROL=y
# 32.768 KHz low speed external clock
CONFIG_CLOCK_STM32_LSE=y
# 8 MHz high speed external clock
CONFIG_CLOCK_STM32_HSE_CLOCK=8000000
# Select HSE for the PLL Source Mux
CONFIG_CLOCK_STM32_PLL_SRC_HSE=y
# PLLM output divisor (8000000 MHz)
CONFIG_CLOCK_STM32_PLL_M_DIVISOR=1
# PLLN output multiplier (320000000 MHz)
CONFIG_CLOCK_STM32_PLL_N_MULTIPLIER=40
# PLLR output divider (80000000 MHz)
CONFIG_CLOCK_STM32_PLL_R_DIVISOR=4
# Unused
CONFIG_CLOCK_STM32_PLL_Q_DIVISOR=2
# Unused
CONFIG_CLOCK_STM32_PLL_P_DIVISOR=2

arewhyan, 2019-04-12:
What debug options are necessary to debug Zephyr on STM32? These are the only options I have set.

set *0xE0042004 = (*0xE0042004) | 0x7
set *0xE0042008 = (*0xE0042008) | 0x1000

Any breakpoint I set prior to mcuboot main, causes the debugger (TrueSTUDIO) to exit. I'm able to debug MCUboot code fine.

erwango, 2019-11-21:
@arewhyan sorry for getting back to you this late on this issue.

Is your question still valid ?

psidhu, 2019-11-21:
I literally (yesterday) ran into this as well. Basically, for me, my system runs at 80MHz. I want to run the SPI at 40MHz, but it hangs. What happens is that the gpio speed is too slow in the default pinmux which causes the RXE to never fire (and thus we hang). 

Can you change your SCK, MISO, MOSI to the following: (STM32_PINMUX_ALT_FUNC_5 | STM32_PUSHPULL_NOPULL | STM32_OSPEEDR_VERY_HIGH_SPEED)

For me, that works. @erwango Depending on what @arewhyan reports, I can push this change.

erwango, 2019-11-22:
@psidhu, thanks for the input!

This looks like what was done on F4 some time ago. At that time we thought it was a F4 limitation.
Though, changing SCK speed to vey high was sufficient. Can you try with only this change ?

psidhu, 2019-11-22:
@erwango I originally made the rest very high because I figured if it's too slow for the clock, at least MOSI would be too slow to transition as well.

I did what you suggested, but communications fail as I thought they would (at least at 40MHz SPI), but no hang. However, like I thought, setting very high for only SCK and MOSI works for both issues.

I also had to make sure that SCK was pulled up (CPOL/CPHA), else I saw transfer errors with multiple devices. I think we should have a way to dynamically set the pull for this pin, but it might have to wait until we get pinmux into devicetree.

For me:
```
#define STM32L4X_PINMUX_FUNC_PB13_SPI2_SCK   \
	(STM32_PINMUX_ALT_FUNC_5 | STM32_PUSHPULL_PULLUP | STM32_OSPEEDR_VERY_HIGH_SPEED)
#define STM32L4X_PINMUX_FUNC_PB14_SPI2_MISO  \
	(STM32_PINMUX_ALT_FUNC_5 | STM32_PUPDR_PULL_DOWN)
#define STM32L4X_PINMUX_FUNC_PB15_SPI2_MOSI  \
	(STM32_PINMUX_ALT_FUNC_5 | STM32_PUPDR_PULL_DOWN | STM32_OSPEEDR_VERY_HIGH_SPEED)
```

erwango, 2019-11-26:
> I think we should have a way to dynamically set the pull for this pin, but it might have to wait until we get pinmux into devicetree.

@psidhu, agreed.
 @arewhyan, can you confirm this is actually solving your issue?

psidhu, 2020-01-15:
@erwango Want me to create a PR for this?

erwango, 2020-01-16:
@psidhu, yes please

PR #21982 (2020-01-17): pinmux: stm32: l4: Make SPI SCK and MOSI pins Very High Speed
Ensure that the SPI SCK and MOSI pins are set to very high as slower
SPI speeds will cause device hangs when used. This is due to the state
changing too fast for hardware, causing RXE to never fire and thus
hang.

Fixes #15435
closed 2020-01-17 by MaureenHelm