zephyrproject-rtos/zephyr #15444: Error initiating sdhc disk

lucaspeixotot, 2019-04-14:
I tried to run samples/subsys/fs and I got error in `disk_access_init()` function. Tracking the problem, I noticed that the error was at the beginner of the communication between mcu and sdhc, when the mcu needs to send some CMD messages to initiate the communication. The error was, more specifically, in the sending of CMD9(SDHC_SEND_CSD) message.

```
(file disk_accesss_sdhc.c, sdhc_detect function, line 754)
/* Read the CSD */
	err = sdhc_cmd_r1(data, SDHC_SEND_CSD, 0);
	if (err != 0) {
		return err;
	}

	err = sdhc_rx_block(data, buf, sizeof(buf));
	if (err != 0) {
		return err;
	}

```

If we look `sdhc_cmd_r1_raw()` function, which is called inside `sdhc_cmd_r1()` function, we will see that we are discarding a byte with this code

```
(file disk_accesss_sdhc.c, sdhc_cmd_r1_raw function, line 457)
/* Ensure there's a idle byte between commands */
	sdhc_rx_u8(data);
```

I think that this is the problem, because according to this table we can't discard every time,

[image]

there is some commands that have a data packet response after R1, R3/R7 response, so the existence of the code above is eating the message token of data packet response.

To solve this bug I just replace the code above with this if statement

```
	if (cmd != SDHC_SEND_CSD && cmd != SDHC_SEND_CID &&
	    cmd != SDHC_READ_SINGLE_BLOCK && cmd != SDHC_READ_MULTIPLE_BLOCK &&
	    cmd != SDHC_WRITE_BLOCK && cmd != SDHC_WRITE_MULTIPLE_BLOCK) {
		sdhc_rx_u8(data);
	}
```

and the sdhc disk was initiated and mounted successfully.

Pathak94, 2019-04-30:
Thank you for the fix! I also encountered this issue with nrf52840_blip board, by doing the fix you suggested, the problem is solved. Will you send a PR for this?

lucaspeixotot, 2019-04-30:
I'm glad it worked for you. Probably this week I will create a PR.

As you are using sdhc worth another tip, the unmount function was not implemented for fatfs. So I created a PR that you can see here #15712

Pathak94, 2019-05-02:
Noted! Thanks.

PR #17700 (2019-08-01): disk: fixing the sending of commands with r1 response
**NOTE** Rebased version of #15790 

The sdhc_cmd_r1_raw() function doesn't take into account the existence
of commands with data response. Because of this, some datas were being
lost.

The commands that return a r1 response and a data are: SDHC_SEND_CSD,
SDHC_SEND_CID, SDHC_READ_SINGLE_BLOCK, SDHC_READ_MULTIPLE_BLOCK,
SDHC_WRITE_BLOCK, SDHC_WRITE_MULTIPLE_BLOCK.

In order to solve this, was juts necessary skip the byte discard when
the command is one of these.

This problem was affecting, for example, the sdhc initialization. The
token ret [... cut]
commit e3f2add6 (2019-08-01): disk: fixing the sending of commands with r1 response

The sdhc_cmd_r1_raw() function doesn't take into account the existence
of commands with data response. Because of this, some datas were being
lost.

The commands that return a r1 response and a data are: SDHC_SEND_CSD,
SDHC_SEND_CID, SDHC_READ_SINGLE_BLOCK, SDHC_READ_MULTIPLE_BLOCK,
SDHC_WRITE_BLOCK, SDHC_WRITE_MULTIPLE_BLOCK.

In order to solve this, was juts necessary skip the byte discard when
the command is one of these.

This problem was affecting, for example, the sdhc initialization. The
token returned from SDHC_SEND_CSD was being l [... cut]
PR #17955 (2019-09-25): [Backport v1.14-branch] disk: fixing the sending of commands with r1 response
The sdhc_cmd_r1_raw() function doesn't take into account the existence
of commands with data response. Because of this, some datas were being
lost.

The commands that return a r1 response and a data are: SDHC_SEND_CSD,
SDHC_SEND_CID, SDHC_READ_SINGLE_BLOCK, SDHC_READ_MULTIPLE_BLOCK,
SDHC_WRITE_BLOCK, SDHC_WRITE_MULTIPLE_BLOCK.

In order to solve this, was juts necessary skip the byte discard when
the command is one of these.

This problem was affecting, for example, the sdhc initialization. The
token returned from SD [... cut]
commit e98cb34c (2019-09-25): disk: fixing the sending of commands with r1 response

The sdhc_cmd_r1_raw() function doesn't take into account the existence
of commands with data response. Because of this, some datas were being
lost.

The commands that return a r1 response and a data are: SDHC_SEND_CSD,
SDHC_SEND_CID, SDHC_READ_SINGLE_BLOCK, SDHC_READ_MULTIPLE_BLOCK,
SDHC_WRITE_BLOCK, SDHC_WRITE_MULTIPLE_BLOCK.

In order to solve this, was juts necessary skip the byte discard when
the command is one of these.

This problem was affecting, for example, the sdhc initialization. The
token returned from SDHC_SEND_CSD was being l [... cut]
closed 2019-08-01 by galak