zephyrproject-rtos/zephyr #15448: help to use zephyr-ninja flash of st_nucleo_l476rg

chaos4284, 2019-04-15:
I can cmake the project sample/hello_world, then when i flash to the board the error open:

[0/1] Flashing nucleo_l476rg
Using runner: openocd
Open On-Chip Debugger 0.10.0+dev-g0b1cbf175-dirty (2019-03-08-20:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : clock speed 500 kHz
Error: open failed

**ERROR: command exited with status 1: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/zephyrproject/zephyr/boards/arm/nucleo_l476rg/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset halt' -c 'verify_image /home/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset run' -c shutdown**

run as "west -v flash --skip-rebuild" for a stack trace
FAILED: zephyr/cmake/flash/CMakeFiles/flash
cd /home/zephyrproject/zephyr/samples/hello_world/build && /home/kwon/cmake-3.13.1-Linux-x86_64/bin/cmake -E env /usr/local/bin/west flash --skip-rebuild
ninja: build stopped: subcommand failed.

Can you find a solution to this problem?

mbolivar, 2019-04-16:
Hi @chaos4284 and thanks for the report.

```
My system is WUL(Ubuntu 18.04) of Windows 10.
```

I am not 100% sure, but I don't think you can flash devices from the WSL command line, as the `/dev` tree there doesn't mirror the windows device tree completely.

Can you try building and flashing using an installation built with cmd.exe instead? The getting started guide (https://docs.zephyrproject.org/latest/getting_started/index.html) has a windows sub-page with the details.

If that works, then it's an issue with WSL and flashing, not Zephyr's build and flash code.

chaos4284, 2019-04-17:
In use cmd.exe 
Occur still problem .
[0/1] Flashing nucleo_l476rg
Traceback (most recent call last):
  File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python37\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 499, in main
    wrap(wrap_argv)
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 485, in wrap
    west.main.main(argv)
  File "C:\zephyrproject\.west\west\src\west\main.py", line 580, in main
    args.handler(args, unknown)
  File "C:\zephyrproject\.west\west\src\west\main.py", line 332, in ext_command_handler
    command.run(*west_parser.parse_known_args(argv))
  File "C:\zephyrproject\.west\west\src\west\commands\command.py", line 90, in run
    self.do_run(args, unknown)
  File "c:\zephyrproject\zephyr\scripts\west_commands\flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "C:\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 151, in do_run_common
    build_dir = _build_dir(args)
  File "C:\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 135, in _build_dir
    elif is_zephyr_build(cwd):
  File "C:\zephyrproject\.west\west\src\west\build.py", line 31, in is_zephyr_build
    cache = cmake.CMakeCache.from_build_dir(path)
  File "C:\zephyrproject\.west\west\src\west\cmake.py", line 166, in from_build_dir
    return CMakeCache(os.path.join(build_dir, DEFAULT_CACHE))
  File "C:\zephyrproject\.west\west\src\west\cmake.py", line 170, in __init__
    self.load(cache_file)
  File "C:\zephyrproject\.west\west\src\west\cmake.py", line 175, in load
    for line_no, line in enumerate(cache):
UnicodeDecodeError: 'cp949' codec can't decode byte 0xe2 in position 1237: illegal multibyte sequence
FAILED: zephyr/cmake/flash/CMakeFiles/flash
cmd.exe /C "cd /D C:\zephyrproject\zephyr\samples\hello_world\build && "C:\Program Files\CMake\bin\cmake.exe" -E env C:/Python37/Scripts/west.exe flash --skip-rebuild"
ninja: build stopped: subcommand failed.

carlescufi, 2019-04-17:
@chaos4284 please install OpenOCD for Windows and place it in your PATH.

chaos4284, 2019-04-17:
expected to be reason for firewall, close this question.

mbolivar, 2019-04-18:
> this issue was confirmed to be caused by a firewall.
> I am going to closed this issue after confirming more it.

Thank you for the feedback. If your investigation shows a way we could detect the firewall issue and print a better error message instead of a stack trace, please let us know! Thanks.

chaos4284, 2019-04-19:
Afterward ninja debug, 
    [0/1] Debugging nucleo_l476rg
Using runner: openocd
Open On-Chip Debugger 0.10.0+dev-g0b1cbf175-dirty (2019-03-08-20:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
/opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : clock speed 500 kHz
Error: open failed

ERROR: command exited with status 127: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb -ex 'target remote :3333' /home/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.elf
run as "west -v debug --skip-rebuild" for a stack trace
FAILED: zephyr/cmake/flash/CMakeFiles/debug
cd /home/zephyrproject/zephyr/samples/hello_world/build && /home/kwon/cmake-3.13.1-Linux-x86_64/bin/cmake -E env /usr/local/bin/west debug --skip-rebuild
ninja: build stopped: subcommand failed.

mbolivar, 2019-04-22:
@chaos4284 I guess this means port 3333 is blocked by your firewall; is that right?

If there is a problem with port 3333 but you can use some other port in your firewall, you can use `west debug --gdb-port SOME_OTHER_NUMBER` instead of `ninja debug` to pick a value that will work for you.

mbolivar, 2019-04-30:
> @chaos4284 I guess this means port 3333 is blocked by your firewall; is that right?

Hi @chaos4284, just following up on this.

mbolivar, 2019-05-16:
Closing since the submitter says they've figured it out and hasn't replied in the past month.

closed 2019-05-16 by mbolivar