zephyrproject-rtos/zephyr #15472: DNS resolver sample sends only one A query

pfl, 2019-04-15:
samples/net/dns_resolve/README.md seems to indicate that both A and AAAA queries for www.zephyrproject.org would be sent when IPv4 and IPv6 are enabled. Also there are two DNS entries in the example, one for IPv6 and one for IPv4.

What actually happens is that only one A query is sent to the first DNS server, no AAAA queries are sent nor is the second DNS server used.  MDNS messages are not sent either.

Since the sample code and the README are conflicting, what is the intended outcome of this sample application?

jukkar, 2019-04-16:
Hmm, works for me. Below is capture from wireshark
```
No.     Time           Source                Destination           Protocol Length Info
      1 0.000000000    2001:db8::1           2001:db8::2           DNS      101    Standard query 0x0f94 A www.zephyrproject.org

Frame 1: 101 bytes on wire (808 bits), 101 bytes captured (808 bits) on interface 0
Ethernet II, Src: RealtekU_12:34:56 (52:54:00:12:34:56), Dst: IcannIan_00:53:ff (00:00:5e:00:53:ff)
Internet Protocol Version 6, Src: 2001:db8::1, Dst: 2001:db8::2
User Datagram Protocol, Src Port: 39238, Dst Port: 15353
Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
      2 0.000121585    2001:db8::2           2001:db8::1           DNS      101    Standard query response 0x0f94 Refused A www.zephyrproject.org

Frame 2: 101 bytes on wire (808 bits), 101 bytes captured (808 bits) on interface 0
Ethernet II, Src: IcannIan_00:53:ff (00:00:5e:00:53:ff), Dst: RealtekU_12:34:56 (52:54:00:12:34:56)
Internet Protocol Version 6, Src: 2001:db8::2, Dst: 2001:db8::1
User Datagram Protocol, Src Port: 15353, Dst Port: 39238
Domain Name System (response)

No.     Time           Source                Destination           Protocol Length Info
      3 0.002165741    2001:db8::1           2001:db8::2           DNS      101    Standard query 0xd342 AAAA www.zephyrproject.org

Frame 3: 101 bytes on wire (808 bits), 101 bytes captured (808 bits) on interface 0
Ethernet II, Src: RealtekU_12:34:56 (52:54:00:12:34:56), Dst: IcannIan_00:53:ff (00:00:5e:00:53:ff)
Internet Protocol Version 6, Src: 2001:db8::1, Dst: 2001:db8::2
User Datagram Protocol, Src Port: 39238, Dst Port: 15353
Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
      4 0.002240778    2001:db8::2           2001:db8::1           DNS      101    Standard query response 0xd342 Refused AAAA www.zephyrproject.org

Frame 4: 101 bytes on wire (808 bits), 101 bytes captured (808 bits) on interface 0
Ethernet II, Src: IcannIan_00:53:ff (00:00:5e:00:53:ff), Dst: RealtekU_12:34:56 (52:54:00:12:34:56)
Internet Protocol Version 6, Src: 2001:db8::2, Dst: 2001:db8::1
User Datagram Protocol, Src Port: 15353, Dst Port: 39238
Domain Name System (response)
```

pfl, 2019-04-16:
> No.     Time           Source                Destination           Protocol Length Info
>       3 0.002165741    2001:db8::1           2001:db8::2           DNS      101    Standard query 0xd342 AAAA www.zephyrproject.org

This is the query I don't see, I wonder why. But notice that only the first DNS server is consulted, not the second one. Let me investigate...

pfl, 2019-04-19:
Strange, I still see only one query for A resource record to the IPv6 name server when I run the sample on SAM E70 Xplained.

pfl, 2019-04-19:
> Standard query response 0xd342 Refused AAAA www.zephyrproject.org

That might be the explanation, actually. You seem to get Refused responses for both A and AAAA records, I get a response for the first query, and the dns_resolve application seems to be happy with such a result?

pfl, 2019-04-19:
Seems that DNS over IPv6 and MDNS both get cancelled:
```
[00:00:03.080,000] <inf> net_config: Initializing network
[00:00:03.080,000] <inf> net_config: IPv4 address: 192.168.255.20
[00:00:03.190,000] <inf> net_config: IPv6 address: fe80::fec2:3dff:fe12:f3ab
[00:00:03.190,000] <inf> net_config: IPv6 address: fe80::fec2:3dff:fe12:f3ab
[00:00:04.090,000] <err> net_ipv6: Prefix 2001:db8:ff:1::/64 could not be added to iface 0x20409e80
[00:00:04.200,000] <inf> net_config: IPv6 address: fe80::fec2:3dff:fe12:f3ab
[00:00:07.210,000] <inf> net_dns_resolve_client_sample: Starting DNS resolve sample
[00:00:07.210,000] <inf> net_dns_resolve_client_sample: IPv4 address: 192.168.255.20
[00:00:07.210,000] <dbg> net_dns_resolve_client_sample.setup_ipv4: DNS id 56547
[00:00:07.210,000] <dbg> net_dns_resolve_client_sample.do_ipv6_lookup: DNS id 14006
[00:00:07.320,000] <inf> net_dns_resolve_client_sample: www.zephyrproject.org IPv4 address: 23.185.0.1
[00:00:07.320,000] <inf> net_dns_resolve_client_sample: DNS resolving finished
[00:00:07.320,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv4_lookup: Doing mDNS IPv4 query
[00:00:07.320,000] <err> net_conn: Identical connection handler 0x20400f58 already found.
[00:00:07.320,000] <dbg> net_dns_resolve_client_sample.do_mdns_ipv4_lookup: mDNS id 12134
[00:00:09.220,000] <inf> net_dns_resolve_client_sample: DNS query was canceled
[00:00:09.330,000] <inf> net_dns_resolve_client_sample: DNS query was canceled
```

pfl, 2019-04-19:
In the subsys/net/lib/dns/resolve.c we have the following code, that on a successful DNS response (where items > 0) does the following:
```
	if (k_delayed_work_remaining_get(&ctx->queries[query_idx].timer) > 0) {
		k_delayed_work_cancel(&ctx->queries[query_idx].timer);
	}
```
So all is fine, any queued DNS queries are cancelled, as the log says above. So the README could see some fine tuning, as it is now, the impression can be that both A and AAAA queries are sent to both or all configured DNS servers in prj.conf. Other than that, case closed.

jukkar, 2019-05-22:
> There might still be a bug here.

Please reopen if there is an issue. Anyway, closing for now.

closed 2019-05-22 by jukkar