zephyrproject-rtos/zephyr #15494: 2.0 Release Checklist

galak, 2019-04-16:
Add items as required
--------
- [ ] Major enhancements:

- [ ] Pre RC1 Steps
   - [ ] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [ ] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as unexploitable.

See https://github.com/zephyrproject-rtos/zephyr/wiki/Release-procedure for tagging details

--------
- [ ] Release Notes WIP
  - [ ] Create draft doc/release-notes-2.0.rst with feature summary from https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update doc/release-notes-2.0.rst details sections (relative to 1.14.0 release) as important issues are addressed (subsystem owners)
--------
- [ ] Release data
  - [ ] Update version in VERSION
  - [ ] Run sanitycheck with --all --release and checkin resulting file scripts/sanity_chk/sanity_last_release.csv
--------
- [ ] Finalize Release Notes
  - [ ] Add list of GitHub issues (spell check issue list and fix in GitHub as needed)
  - [ ] Review doc/release-notes/release-notes-2.0.rst overview summary and details sections (see Sign-off below)
  - [ ] Update doc/release-notes/index.rst (add reference to 2.0 notes)
  - [ ] Update doc/index.rst (add reference to doc/2.0.0)
  - [ ] Update doc/conf.py (add version to pick list)
  - [ ] Update doc/LICENSING.rst for new components that do not have a license
  - [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases
  - [ ] Final tagged release as zephyr-v2.0.0 (with release notes)
  - [ ] Update https://www.zephyrproject.org/developers/#releases-overview
  - [ ] Update https://www.zephyrproject.org/developers/#downloads (and git checkout version example)
  - [ ] Update doc generation (in future should be automatic based on tag) ping @nashif
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release
--------
- [ ]  Final Release Notes Sign-off
  - Subsystems
    - [ ] Networking - @jukkar 
    - [ ] Bluetooth - @jhedberg , @carlescufi 
  - Architectures
    - [ ] ARC - @ruuddw 
    - [ ] ARM - @galak, @MaureenHelm 
    - [ ] X86 - @nashif, @andrewboie 
    - [ ] Xtensa, RISCV32, NIOS-II - @nashif, @andrewboie 
  - [ ] Security - ????
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Create branch for 2.0.x releases, update github protections for branch
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Update https://www.zephyrproject.org with latest stats from Github [@dbkinder]
  - [ ] Move all PRs/Issues in the 2.0 milestone to the next milestone
  - [ ] Close v2.0.0 milestone
  - [ ] Create v2.0.1 milestone

ioannisg, 2019-08-10:
@MaureenHelm @nashif @galak @carlescufi I am trying to resolve this pre-RC1 task:
>  Verify that all external components and external dependencies are up to date, for example mbedtls.

However, now such external components are outside the main repository, so I am wondering, if we can skip this checking now. That would make sense, since these components can be updated independently of the main repo (that is, regardless of feature freeze).

Or: does this, now, mean that we need to tag all repositories together with the main zephyr repo?

aescolar, 2019-08-10:
@ioannisg, 2 comments fro me:
* There is still a few ext/ in tree (e.g. tinycrypt, which is up to date)
* I would say that them being in a separate repo does not change things here. As you said, you are tagging zephyr with a west manifest which lists the "known to work with" version of those other repos. So those should be up to date

ioannisg, 2019-08-10:
> 
> 
> @ioannisg, 2 comments from me:
> 
>     * There is still a few ext/ in tree (e.g. tinycrypt, which is up to date)
> 

@aescolar thanks, for your help here :)
Sure, I am aware of the components in the tree; tinucrypt is indeed up-to-date

>     * I would say that them being in a separate repo does not change things here. As you said, you are tagging zephyr with a west manifest which lists the "known to work with" version of those other repos. So those should be up to date

I agree they need to be up-to-date, in general; but does it mean "latest release" or we are ok with "latest major release"?

Second: updating external components (in the standalone repos) and west revisions (in zephyr) will require some testing, righti? if I should do this now, I can't tag -rc1 until this testing is completed. So what level of testing would we need? I suppose green CI should suffice, or?

ioannisg, 2019-09-07:
@nashif @dbkinder there is one box left to tick in the Final Release Notes, could you check if this has been actually done?

dbkinder, 2019-09-07:
V2.0.0 docs are published (thanks @nashif ) completing the last item in the “Finalize release notes” section, so I checked that off.

ioannisg, 2019-09-10:
2.1 Release checklist: https://github.com/zephyrproject-rtos/zephyr/issues/19053

closed 2019-10-03 by ioannisg