zephyrproject-rtos/zephyr #15495: tests/drivers/spi/spi_loopback/peripheral.spi fails to build on several boards

galak, 2019-04-16:
```
particle_xenon            tests/drivers/spi/spi_loopback/peripheral.spi      FAILED: build_error
	see: sanity-out/particle_xenon/tests/drivers/spi/spi_loopback/peripheral.spi/build.log

total complete:    2/  13  15%  failed:    2

particle_boron            tests/drivers/spi/spi_loopback/peripheral.spi      FAILED: build_error
	see: sanity-out/particle_boron/tests/drivers/spi/spi_loopback/peripheral.spi/build.log

total complete:    3/  13  23%  failed:    3

particle_argon            tests/drivers/spi/spi_loopback/peripheral.spi      FAILED: build_error
	see: sanity-out/particle_argon/tests/drivers/spi/spi_loopback/peripheral.spi/build.log

total complete:    7/  13  53%  failed:    7
```

pabigot, 2019-04-16:
There's a fix for this in #15369 if somebody wants to cherry-pick and merge it.

galak, 2019-04-16:
> There's a fix for this in #15369 if somebody wants to cherry-pick and merge it.

Thanks, I'll cherry-pick that out of the PR for 1.14.

galak, 2019-04-16:
Resolved by #15496

closed 2019-04-16 by galak