zephyrproject-rtos/zephyr #15497: USB DFU: STM32: usb dfu mode doesn't work

jli157, 2019-04-16:
**Describe the bug**
Tried the sample `samples/subsys/uwb/dfu` with `mcuboot` on a Nucleo_F429ZI board, strictly following the instructions by the README. However, the DFU function never works: dfu-util can't do any operations like upload, download or query. The master branch is as of 04/16/2019 (the head is db8d32881f5c5a84abefa75ca7bf36368ec0cbfb). 

**To Reproduce**
Steps to reproduce the behavior:
1. build "mcuboot" by specifying the board to "nucleo_f429zi" and enabling "MULTITHREAD". 
2. flash mcuboot to the board and the boot loader can boot up as expected. 
3. build the sample "samples/subsys/usb/dfu" by the following command:
       `west build -b nucleo_f429zi`
4. Sign the zephyr firmware file by following the instructions in `dfu`'s readme. 
5.  Flash the dfu sample to the board by the command 
     `west flash --hex-file zephyr.signed.hex`
6. Boot the board as expected. 
7. Run the "dfu-util" commands as described in the README to test DFU functions.

**Expected behavior**
The dfu-util commands listed in the `README` should work. However, no one works. The outputs from the commands and device logs are attached. 

**Impact**
prevent my project from enabling the DFU function. 

**Environment (please complete the following information):**
 - OS: Ubuntu 16.04
 - Board: NUCLEO-F429ZI
 - Toolchain: Zephyr SDK 0.10.0
 - The master branch as of 04/16/2019 (HEAD: db8d32881f5c5a84abefa75ca7bf36368ec0cbfb) 

**Screenshots or console output**
* Upload command results
```
sudo dfu-util --alt 0 --upload slot0_backup.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Opening DFU capable USB device...
ID 2fe3:0100
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
Determining device status: state = appIDLE, status = 0
Device really in Runtime Mode, send DFU detach request...
Resetting USB...
dfu-util: Lost device after RESET?
```

* Device logs corresponding to the upload command
```
[00:01:11.100,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:11.110,000] <dbg> usb_device.usb_handle_request: ** 0 **
[00:01:11.110,000] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
[00:01:11.120,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
[00:01:11.130,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
[00:01:11.140,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:11.150,000] <dbg> usb_device.usb_handle_request: ** 0 **
[00:01:11.150,000] <dbg> usb_dfu.dfu_custom_handle_req: DFU alternate setting 0
[00:01:11.160,000] <dbg> usb_device.usb_handle_control_transfer: 0] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:11.170,000] <dbg> usb_device.usb_handle_request: ** 1 **
[00:01:11.180,000] <dbg> usb_dfu.dfu_class_handle_req: DFU_GETSTATUS: status 0, state 0
[00:01:11.190,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
[00:01:11.200,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
[00:01:11.450,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:11.460,000] <dbg> usb_device.usb_handle_request: ** 1 **
[00:01:11.460,000] <dbg> usb_dfu.dfu_class_handle_req: DFU_DETACH timeout 1000, state 0
[00:01:11.470,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
```

* Download command results
```
sudo dfu-util --alt 1 --download zephyr/zephyr-signed.bin 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2fe3:0100
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
Determining device status: state = appDETACH, status = 0
Device really in Runtime Mode, send DFU detach request...
dfu-util: error detaching
Resetting USB...
dfu-util: Lost device after RESET?
```

* Device logs corresponding to Download command
```
[00:01:49.560,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:49.570,000] <dbg> usb_device.usb_handle_request: ** 0 **
[00:01:49.580,000] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
[00:01:49.580,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
[00:01:49.590,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
[00:01:49.600,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:49.610,000] <dbg> usb_device.usb_handle_request: ** 0 **
[00:01:49.610,000] <dbg> usb_dfu.dfu_custom_handle_req: DFU alternate setting 0
[00:01:49.620,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:49.640,000] <dbg> usb_device.usb_handle_request: ** 1 **
[00:01:49.640,000] <dbg> usb_dfu.dfu_class_handle_req: DFU_GETSTATUS: status 0, state 1
[00:01:49.650,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
[00:01:49.660,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
[00:01:49.910,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
[00:01:49.920,000] <dbg> usb_device.usb_handle_request: ** 1 **
[00:01:49.920,000] <dbg> usb_dfu.dfu_class_handle_req: DFU_DETACH timeout 1000, state 1
[00:01:49.930,000] <dbg> usb_device.usb_handle_request: Handler Error 1
[00:01:49.940,000] <dbg> usb_device.usb_print_setup: Setup: 21 0 3e8 0 0
[00:01:49.950,000] <dbg> usb_device.usb_handle_control_transfer: usb_handle_request failed
```

jli157, 2019-04-16:
More logs are attached here:
[dfu_logs-2.txt](https://github.com/zephyrproject-rtos/zephyr/files/3086716/dfu_logs-2.txt)

jli157, 2019-04-16:
By the way, the command `dfu-util -l` sometimes works well, sometimes gives wrong info: 

* Normal outputs:
```
sudo dfu-util -v -l
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Found DFU: [2fe3:0100] ver=0011, devnum=72, cfg=1, intf=0, alt=1, name="image-0", serial="0.01"
Found DFU: [2fe3:0100] ver=0011, devnum=72, cfg=1, intf=0, alt=0, name="image-1", serial="0.01"
```

* Bad outputs
```
sudo dfu-util -v -l
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Found Runtime: [2fe3:0100] ver=0011, devnum=74, cfg=1, intf=0, alt=0, name="UNKNOWN", serial="0.01"
```

jli157, 2019-04-16:
Also, I tried the suggested solutions to the issue #14882, no luck to make it work on the nucleo_f429zi board.

jfischer-phytec-iot, 2019-04-17:
@jli157 _Ubuntu 16.04_ native or inside a VM?

jfischer-phytec-iot, 2019-04-17:
@jli157 Can you test it with http://github.com/zephyrproject-rtos/zephyr/pull/9412 (if yes then please rebase it on latest master) ?

jli157, 2019-04-17:
> @jli157 _Ubuntu 16.04_ native or inside a VM?

My Ubuntu is running in a Parallel virtual machine but the USB devices are given to the virtual machine during my development. I guess the timeout parameters for USB on a VM could be a bit different from native ones. However, should we also support development on a VM? You can also tell me what kind of parameters I should adjust to make it work if you suspect VM is the culprit? I'd like to try it. :)

jli157, 2019-04-17:
> @jli157 Can you test it with [http://github.com/zephyrproject-rtos/zephyr/pull/9412](https://github.com/zephyrproject-rtos/zephyr/pull/9412) (if yes then please rebase it on latest master) ?

Sure, I'll give it a try.

jli157, 2019-04-17:
Host logs for downloading. The downloading took almost 1 hour!
```
dfu-util --alt 1 --download zephyr-signed.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2fe3:0100
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
Determining device status: state = appIDLE, status = 0
Device really in Runtime Mode, send DFU detach request...
Resetting USB...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Setting #1 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 64
Copying data from PC to DFU device
Download	[=========================] 100%       141795 bytes
Download done.
state(2) = dfuIDLE, status(0) = No error condition is present
Done!
```

jfischer-phytec-iot, 2019-04-18:
@jli157 About USB DFU attached to a gast behind a VM: I did not have time to analyze it and I do not see it as a bug as long it works on native host.

About speed, can you please revert ea177e785ca32cb97fb41475377890e2afdd1d17 and test it again on NUCLEO-F429ZI?

jli157, 2019-04-25:
@jfischer-phytec-iot Any suggestions to solve the slow flashing issue introduced by ea177e7?

jfischer-phytec-iot, 2019-04-26:
@jli157 as alternative workaround you can set USB_DFU_DNLOAD_POLLTIMEOUT to 256

I consider to revert  ea177e7 at present there is no solution to be sure that a control had success before start erase process.

jli157, 2019-04-26:
> @jli157 as alternative workaround you can set USB_DFU_DNLOAD_POLLTIMEOUT to 256
> 
> I consider to revert [ea177e7](https://github.com/zephyrproject-rtos/zephyr/commit/ea177e785ca32cb97fb41475377890e2afdd1d17), at present there is no solution to be sure that a control stage had success before start erase process.

Reverting ea177e7 already works for me. I'm just thinking what's the next you like to give it a better solution. 

Do you recommend to close this issue or rename it to something like slowing DFU issue?

jfischer-phytec-iot, 2019-04-26:
Please let it open, I have an idea how to fix but tend to revert it for now.

jli157, 2019-05-17:
@carlescufi and @jfischer-phytec-iot can you reopen this issue? I verified that the issue happened again on the latest master ( fc3270).  Even the command `dfu-util -l` can't list correct partition table. It seems the dfu mode is not very stable. 

My test is on a real Ubuntu 16.04 Linux instead of a VM. Thank you!

jfischer-phytec-iot, 2019-05-17:
I also noticed several issues on the current master, on only with dfu class. Please also note, that some parts of the usb code has been changed. Can you please test `sudo dfu-util -l` on the latest master?

carlescufi, 2019-05-22:
@jli157 can you please retest this with the latest master?

jli157, 2019-05-30:
@carlescufi @jfischer-phytec-iot checked again on the same board. It works well now: I can list out two partitions and download an image file to the board. Thank you!

closed 2019-05-30 by jli157

yairgd, 2021-05-08:
> **Describe the bug**
> Tried the sample `samples/subsys/usb/dfu` with `mcuboot` on a Nucleo_F429ZI board, strictly following the instructions by the README. However, the DFU function never works: dfu-util can't do any operations like upload, download or query. The master branch is as of 04/16/2019 (the head is [db8d328](https://github.com/zephyrproject-rtos/zephyr/commit/db8d32881f5c5a84abefa75ca7bf36368ec0cbfb)).
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> 1. build "mcuboot" by specifying the board to "nucleo_f429zi" and enabling "MULTITHREAD".
> 2. flash mcuboot to the board and the boot loader can boot up as expected.
> 3. build the sample "samples/subsys/usb/dfu" by the following command:
>    `west build -b nucleo_f429zi`
> 4. Sign the zephyr firmware file by following the instructions in `dfu`'s readme.
> 5. Flash the dfu sample to the board by the command
>    `west flash --hex-file zephyr.signed.hex`
> 6. Boot the board as expected.
> 7. Run the "dfu-util" commands as described in the README to test DFU functions.
> 
> **Expected behavior**
> The dfu-util commands listed in the `README` should work. However, no one works. The outputs from the commands and device logs are attached.
> 
> **Impact**
> prevent my project from enabling the DFU function.
> 
> **Environment (please complete the following information):**
> 
> * OS: Ubuntu 16.04 running on a Parallel VM.
> * Board: NUCLEO-F429ZI
> * Toolchain: Zephyr SDK 0.10.0
> * The master branch as of 04/16/2019 (HEAD: [db8d328](https://github.com/zephyrproject-rtos/zephyr/commit/db8d32881f5c5a84abefa75ca7bf36368ec0cbfb))
> 
> **Screenshots or console output**
> 
> * Upload command results
> 
> ```
> sudo dfu-util --alt 0 --upload slot0_backup.bin
> dfu-util 0.8
> 
> Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
> Copyright 2010-2014 Tormod Volden and Stefan Schmidt
> This program is Free Software and has ABSOLUTELY NO WARRANTY
> Please report bugs to dfu-util@lists.gnumonks.org
> 
> Opening DFU capable USB device...
> ID 2fe3:0100
> Run-time device DFU version 0110
> Claiming USB DFU Runtime Interface...
> Determining device status: state = appIDLE, status = 0
> Device really in Runtime Mode, send DFU detach request...
> Resetting USB...
> dfu-util: Lost device after RESET?
> ```
> 
> * Device logs corresponding to the upload command
> 
> ```
> [00:01:11.100,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:11.110,000] <dbg> usb_device.usb_handle_request: ** 0 **
> [00:01:11.110,000] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
> [00:01:11.120,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
> [00:01:11.130,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
> [00:01:11.140,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:11.150,000] <dbg> usb_device.usb_handle_request: ** 0 **
> [00:01:11.150,000] <dbg> usb_dfu.dfu_custom_handle_req: DFU alternate setting 0
> [00:01:11.160,000] <dbg> usb_device.usb_handle_control_transfer: 0] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:11.170,000] <dbg> usb_device.usb_handle_request: ** 1 **
> [00:01:11.180,000] <dbg> usb_dfu.dfu_class_handle_req: DFU_GETSTATUS: status 0, state 0
> [00:01:11.190,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
> [00:01:11.200,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
> [00:01:11.450,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:11.460,000] <dbg> usb_device.usb_handle_request: ** 1 **
[... 28 lines cut ...]
> * Device logs corresponding to Download command
> 
> ```
> [00:01:49.560,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:49.570,000] <dbg> usb_device.usb_handle_request: ** 0 **
> [00:01:49.580,000] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
> [00:01:49.580,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
> [00:01:49.590,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
> [00:01:49.600,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:49.610,000] <dbg> usb_device.usb_handle_request: ** 0 **
> [00:01:49.610,000] <dbg> usb_dfu.dfu_custom_handle_req: DFU alternate setting 0
> [00:01:49.620,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:49.640,000] <dbg> usb_device.usb_handle_request: ** 1 **
> [00:01:49.640,000] <dbg> usb_dfu.dfu_class_handle_req: DFU_GETSTATUS: status 0, state 1
> [00:01:49.650,000] <dbg> usb_device.usb_handle_control_transfer: ep 80, status 2
> [00:01:49.660,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 1
> [00:01:49.910,000] <dbg> usb_device.usb_handle_control_transfer: ep 0, status 0
> [00:01:49.920,000] <dbg> usb_device.usb_handle_request: ** 1 **
> [00:01:49.920,000] <dbg> usb_dfu.dfu_class_handle_req: DFU_DETACH timeout 1000, state 1
> [00:01:49.930,000] <dbg> usb_device.usb_handle_request: Handler Error 1
> [00:01:49.940,000] <dbg> usb_device.usb_print_setup: Setup: 21 0 3e8 0 0
> [00:01:49.950,000] <dbg> usb_device.usb_handle_control_transfer: usb_handle_request failed
> ```

I also has issues with time out, apparently I found the all massive debug log cause the time problems.  Now it work also in virtual linux machine also.

yairgd, 2021-05-08:
I also has issues with time out, apparently I found the all massive debug log cause the time out problems.  Now it work also in virtual Linux machine also.

PR #15747 (2019-05-07): usb: usb_dfu: do not set bwPollTimeout dynamically
Partially revert commit ea177e785ca3
("usb: dfu: set bwPollTimeout dynamically")

Introduced fix does not work proper because there is no way to be
sure that a control stage had success before start erase process.
Instead IMG_ERASE_PROGRESSIVELY configuration should be used
if the erase of the flash takes longer time.

resolves: #15497
commit 53806186 (2019-05-07): usb: usb_dfu: do not set bwPollTimeout dynamically

Partially revert commit ea177e785ca3
("usb: dfu: set bwPollTimeout dynamically")

Introduced fix does not work proper because there is no way to be
sure that a control stage had success before start erase process.
Instead IMG_ERASE_PROGRESSIVELY configuration should be used
if the erase of the flash takes longer time.

resolves: #15497
commit 671dd5c8 (2019-05-07): usb: usb_dfu: do not set bwPollTimeout dynamically

Partially revert commit ea177e785ca3
("usb: dfu: set bwPollTimeout dynamically")

Introduced fix does not work proper because there is no way to be
sure that a control stage had success before start erase process.
Instead IMG_ERASE_PROGRESSIVELY configuration should be used
if the erase of the flash takes longer time.

resolves: #15497
commit 42bdcccd (2019-05-07): usb: usb_dfu: do not set bwPollTimeout dynamically

Partially revert commit ea177e785ca3
("usb: dfu: set bwPollTimeout dynamically")

Introduced fix does not work proper because there is no way to be
sure that a control stage had success before start erase process.
Instead IMG_ERASE_PROGRESSIVELY configuration should be used
if the erase of the flash takes longer time.

resolves: #15497