zephyrproject-rtos/zephyr #15499: gpio_intel_apl: gpio_pin_read() pin value doesn't match documentation

gnuless, 2019-04-16:
The Intel GPIO driver for the Apollo Lake SoC has behavior that is slightly different from the documented interface: `gpio_pin_read()`  (implemented by `gpio_intel_apl_read`) sets `value` to `0` for a low level (as expected), but `2` for a high level.  This is correct as far as C's truth values are concerned, but the documentation for `gpio_pin_read()` says that `value` will be either `0` or `1`.

It's simple enough to fix `gpio_intel_apl.c` to match the documentation, or relax the API itself and update the documentation accordingly.  I'm not sure which fix is better.

gnuless, 2019-04-16:
@dcpleung @andyross @nashif @andrewboie Any thoughts?

dcpleung, 2019-04-17:
Probably missed a bit shift somewhere... if API asks for 0 or 1, we should only return 0 or 1.

gnuless, 2019-04-17:
@dcpleung Yeah, it's a simple fix (the receive state for the GPIO state is in bit 1 of the register, not bit 0).

Ok, I'll implement that fix, rather than changing the API.

PR #15544 (2019-04-18): drivers/gpio_intel_apl.c: fix return value for gpio_pin_read()
gpio_intel_apl_read() should set *value to 1, not 2, when the
GPIO input is a logical high.

Fixes: #15499
commit 8905b0fe (2019-04-18): drivers/gpio_intel_apl.c: fix return value for gpio_pin_read()

gpio_intel_apl_read() should set *value to 1, not 2, when the
GPIO input is a logical high.

Fixes: #15499
closed 2019-04-18 by nashif