zephyrproject-rtos/zephyr #15501: smp_svr build issue

bublover, 2019-04-17:
**Describe the bug**
I was trying to build project with mcugmr enabled.
But error happened.
```
In file included from zephyr/samples/subsys/mgmt/mcumgr/smp_svr/src/main.c:12:
zephyr/include/mgmt/buf.h:11:10: fatal error: cbor_encoder_writer.h: No such file or directory
 #include "cbor_encoder_writer.h"
```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=96b_carbon_nrf51 ..
3. make
4. See error

**Expected behavior**
Build OK

**Impact**
I'd like to use mcumgr function to flash kernel image.
I fristly enabled CONFIG_MCUMGR and CONFIG_MCUMGR_CMD_IMG_MGMT. 
And then built. Error happened.
```
In file included from /home/local/SPREADTRUM/bub.wei/zephyr_uwp5661/zephyr/ext/lib/mgmt/mcumgr/cborattr/src/cborattr.c:21:0:
/home/local/SPREADTRUM/bub.wei/zephyr_uwp5661/zephyr/ext/lib/mgmt/mcumgr/cborattr/include/cborattr/cborattr.h:29:10: fatal error: cbor.h: No such file or directory
 #include "cbor.h"
          ^~~~~~~~
compilation terminated.
```
After this, I trired to build sample smp_svr. Error came again as above.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.2 LTS
 - Toolchain Zephyr SDK 0.10.0

carlescufi, 2019-04-17:
Please follow the instructions on the Getting Started Guide to clone the repo with west and then run:
`west update`. That will pull the tinycbor repo, which seems to be missing from your copy.

bublover, 2019-04-18:
> Please follow the instructions on the Getting Started Guide to clone the repo with west and then run:
> `west update`. That will pull the tinycbor repo, which seems to be missing from your copy.

Hi Carles @carlescufi, as guide, I have done 'pip3 install --user west '    
```
09:41:44<~>$ pip3 install west
Collecting west
  Using cached https://files.pythonhosted.org/packages/e5/9f/7dcd8985532e9b42d76c0777bbfa411b58991f86e82e9d2a0c97948b26e1/west-0.5.8-py3-none-any.whl
Collecting pykwalify (from west)
  Using cached https://files.pythonhosted.org/packages/36/9f/612de8ca540bd24d604f544248c4c46e9db76f6ea5eb75fb4244da6ebbf0/pykwalify-1.7.0-py2.py3-none-any.whl
Collecting configobj (from west)
Collecting colorama (from west)
  Using cached https://files.pythonhosted.org/packages/4f/a6/728666f39bfff1719fc94c481890b2106837da9318031f71a8424b662e12/colorama-0.4.1-py2.py3-none-any.whl
Collecting PyYAML (from west)
Collecting docopt>=0.6.2 (from pykwalify->west)
Collecting python-dateutil>=2.4.2 (from pykwalify->west)
  Using cached https://files.pythonhosted.org/packages/41/17/c62faccbfbd163c7f57f3844689e3a78bae1f403648a6afb1d0866d87fbb/python_dateutil-2.8.0-py2.py3-none-any.whl
Collecting six (from configobj->west)
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Installing collected packages: PyYAML, docopt, six, python-dateutil, pykwalify, configobj, colorama, west
Successfully installed PyYAML-5.1 colorama-0.4.1 configobj-5.0.6 docopt-0.6.2 pykwalify-1.7.0 python-dateutil-2.8.0 six-1.12.0 west-0.5.8
```
But No positive response from 'west --version'.
```
09:42:36<~>$ west --version
bash: west: unknow command
```

carlescufi, 2019-04-18:
This sounds like your Python installation has not added the folder where it stores the executables  to the PATH. Please check that `~/.local/bin` is in your PATH

bublover, 2019-04-18:
> This sounds like your Python installation has not added the folder where it stores the executables to the PATH. Please check that `~/.local/bin` is in your PATH

Thx @carlescufi , working.   BTW, could you pls tell me how to use mcumgr to download image via UART?

carlescufi, 2019-04-18:
@bublover 

You need to use a serial port profile connection with mcumgr, something like:

`mcumgr conn add ttyACM0 connstring='dev=/dev/ttyACM0,baud=115200,mtu=512'`
and then
`mcumgr -c ttyACM0 image list`
and so on

closed 2019-04-18 by carlescufi