zephyrproject-rtos/zephyr #15507: NRF52840: usb composite MSC + HID (with CONFIG_ENABLE_HID_INT_OUT_EP)

dmazzella, 2019-04-17:
**Describe the bug**
Composite usb device MSC + HID does not work on Windows, the device does not complete the installation.
MSC can be either configured with either one of `CONFIG_DISK_ACCESS_RAM` or `CONFIG_DISK_ACCESS_FLASH`, HID has `CONFIG_ENABLE_HID_INT_OUT_EP=y` 

**To Reproduce**
1. git clone https://github.com/dmazzella/zephyr.git
2. cd zephyr/samples/subsys/usb/hid_w_out_ep-mass/
3. mkdir build && cd build
4. cmake -GNinja -DBOARD=nrf52840_pca10059 ..
5. ninja
6. flash zephyr.hex with Nrf Connect -> Programmer

**Expected behavior**
After writing the firmware using nRF Connect Programmer, I expect a composite  MSC + HID device (PCA10059) to be properly recognized. 
This is the case on Linux and macOS but on Windows 10 the USB composite device appears as _unknown device_.

**Impact**
Showstopper.

**Screenshots or console output**
[image]
[image]

**Environment (please complete the following information):**
 - OS:
    - Built on Windows 10 WSL, uname output `Linux dma-pc 4.4.0-17763-Microsoft #379-Microsoft x86_64 x86_64 x86_64 GNU/Linux`
    - Tested on Windows 10 - Version 1809 (build SO 17763.437)
 - Toolchain:
    - Zephyr version: 1.14.99
 - Commit SHA or Version used:
    - zephyrproject-rtos/zephyr 47b7c79e39365065876207d9f28593182cc40029

**Additional context**
I can try only with NRF52840 PCA10059

anddam, 2019-04-17:
I figure #15486 can be closed now.

finikorg, 2019-04-17:
I tested on `reel_board` sample `samples/subsys/usb/hid` with enabled MSC, works fine.

dmazzella, 2019-04-17:
yes, it work. The problem is triggered by enabling `CONFIG_ENABLE_HID_INT_OUT_EP=y`

finikorg, 2019-04-18:
> yes, it work. The problem is triggered by enabling `CONFIG_ENABLE_HID_INT_OUT_EP=y`

It works for me on `reel_board` with enabled `CONFIG_ENABLE_HID_INT_OUT_EP=y`

dmazzella, 2019-04-18:
can you please paste your .conf used for test?

this is mine for ```nrf52840_pca10059``` with sample  ```samples/subsys/usb/hid```

```
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_COMPOSITE_DEVICE=y
CONFIG_USB_DEVICE_HID=y
CONFIG_ENABLE_HID_INT_OUT_EP=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr HID sample"
CONFIG_USB_HID_BOOT_PROTOCOL=y

CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
CONFIG_DISK_ACCESS_RAM=y
CONFIG_MASS_STORAGE_DISK_NAME="RAM"

CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
```

without `CONFIG_ENABLE_HID_INT_OUT_EP=y` work fine for me.

dmazzella, 2019-04-18:
@finikorg @masz-nordic none of you have the chance to try with `nrf52840_pca10059`?

masz-nordic, 2019-04-18:
I was able to reproduce with `nrf52840_pca10059`.
`GET_DESCRIPTOR(Configuration)` gets stalled. Its size is 64 bytes (which is equal to the size of an IN transaction). Probably a zero length packet should be added.

dmazzella, 2019-04-19:
@masz-nordic can i try something?

finikorg, 2019-04-19:
@dmazzella to verify that issue @masz-nordic mentioned you can add another HID, it will increase descriptor table.

dmazzella, 2019-04-19:
@masz-nordic @finikorg I tried adding another HID and the device on Windows works as expected.
Is a fix without adding a second HID possible?

finikorg, 2019-04-20:
> @masz-nordic @finikorg I tried adding another HID and the device on Windows works as expected.
> Is a fix without adding a second HID possible?

This should be fixed for the board device controller, I think this is serious issue, since all control packets multiple of 64 bytes fails, according to this finding.

dmazzella, 2019-04-28:
@finikorg i have tried with commit d44086db5a2aa362510d7ac1a1f2f401c8ff180c and now on macOS I have this error:

```bash
MBP-di-Damiano:~$ sudo dmesg
Command 0x1100 completed. inCount: 2 data:0x0Command 0x2 completed. inCount: 2 data:0x4Command 0x1 completed. inCount: 2 data:0x1101Command 0x16 completed. inCount: 2 data:0xc0Command 0xa completed. inCount: 2 data:0xfc35Command 0xb completed. inCount: 2 data:0xfba4Command 0x12 completed. inCount: 2 data:0x5bCommand 0x13 completed. inCount: 2 data:0xffffCommand 0xf completed. inCount: 2 data:0x698Command 0x10 completed. inCount: 2 data:0x13d5SmartBattery: finished polling type 4
USBMSC Identifier (non-unique): ABCDEF012345 0x2fe3 0x100 0x11, 1
Report bytes: input:2 output:2 feature:1Collections : 2Buttons     : input:0 output:0 feature:0Values      : input:1 output:1 feature:0Element value capacity 96Report count: 3Report ID: 0 input:0 output:0 feature:0Report ID: 1 input:16 output:8 feature:8Report ID: 2 input:8 output:16 feature:8USB device 2FE3010014300000 - will be reset!
005288.316806 IOUSBHostHIDDevice: IOUSBHostHIDDevice::interruptReadComplete: IO error 0xe00002ed, 9 retries remaining
005290.163284 AppleUSBHostCompositeDevice@: AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe0005000)
USB device 2FE3010014300000 - failed to re-instantiate!
USB device 2FE3010014300000 - USB Mass Storage stack will be terminated!
USB device 2FE3010014300000 - fConsecutiveResetCount = 1.
MBP-di-Damiano:~$
```

with .conf:

```
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_COMPOSITE_DEVICE=y
CONFIG_USB_DEVICE_HID=y
CONFIG_ENABLE_HID_INT_OUT_EP=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr HID sample"
CONFIG_USB_HID_BOOT_PROTOCOL=y

CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
CONFIG_DISK_ACCESS_RAM=y
CONFIG_MASS_STORAGE_DISK_NAME="RAM"

CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
```

dmazzella, 2019-04-28:
@finikorg `nrf52840_pca10059` with sample `samples/subsys/usb/mass` my .conf:

```
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr MSC sample"
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
CONFIG_DISK_ACCESS_RAM=y
CONFIG_MASS_STORAGE_DISK_NAME="RAM"

```

error: 
```bash
MBP-di-Damiano:~$ sudo dmesg
USBMSC Identifier (non-unique): ABCDEF012345 0x2fe3 0x100 0x11, 1
USB device 2FE3010014300000 - will be reset!
006311.172343 AppleUSBHostCompositeDevice@: AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe0005000)
USB device 2FE3010014300000 - failed to re-instantiate!
USB device 2FE3010014300000 - USB Mass Storage stack will be terminated!
USB device 2FE3010014300000 - fConsecutiveResetCount = 1.
MBP-di-Damiano:~$ 

```

finikorg, 2019-04-29:
@dmazzella So does it work in Windows?

dmazzella, 2019-04-29:
> @dmazzella So does it work in Windows?

 `samples/subsys/usb/mass` work on Windows. YES

`samples/subsys/usb/hid` with conf: 

```
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_COMPOSITE_DEVICE=y
CONFIG_USB_DEVICE_HID=y
CONFIG_ENABLE_HID_INT_OUT_EP=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr HID sample"
CONFIG_USB_HID_BOOT_PROTOCOL=y

CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
CONFIG_DISK_ACCESS_RAM=y
CONFIG_MASS_STORAGE_DISK_NAME="RAM"

CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
```

YES

dmazzella, 2019-04-30:
@finikorg `nrf52840_pca10059` with samples `samples/subsys/usb/cdc_acm` and `samples/subsys/usb/cdc_acm_composite`  they don't work anymore

finikorg, 2019-04-30:
```
--- a/subsys/usb/usb_descriptor.h
+++ b/subsys/usb/usb_descriptor.h
@@ -34,8 +34,8 @@
 #define USB_STRING_DESCRIPTOR_LENGTH(s)        (sizeof(s) * 2)
 
 /* Automatic endpoint assignment */
-#define AUTO_EP_IN                     0x81
-#define AUTO_EP_OUT                    0x01
+#define AUTO_EP_IN                     0x80
+#define AUTO_EP_OUT                    0x00
```

Does this help?

dmazzella, 2019-04-30:
> --- a/subsys/usb/usb_descriptor.h
> +++ b/subsys/usb/usb_descriptor.h
> @@ -34,8 +34,8 @@
>  #define USB_STRING_DESCRIPTOR_LENGTH(s)        (sizeof(s) * 2)
>
>  /* Automatic endpoint assignment */
> -#define [...]

Yes now work correctly

dmazzella, 2019-04-30:
@finikorg sorry if I said that now it works without having tested on all three operative systems.

- _Windows_: work correctly
- _MacOS_: still does not work
- _Linux_: untested

dmazzella, 2019-05-04:
@finikorg @masz-nordic any progress on this issue?

finikorg, 2019-05-07:
> @finikorg @masz-nordic any progress on this issue?

@dmazzella So what is the issue now? `samples/subsys/usb/mass` does not work on Mac? In all configurations or only with `OUT_EP`?

dmazzella, 2019-05-07:
@finikorg `samples/subsys/usb/mass` does not work on Mac in all configurations.

i have used `nrf52840_pca10059` with .conf:

```ini
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr MSC sample"
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
CONFIG_DISK_ACCESS_RAM=y
CONFIG_MASS_STORAGE_DISK_NAME="RAM"
```

```bash
MBP-di-Damiano:~$ sudo dmesg
USBMSC Identifier (non-unique): ABCDEF012345 0x2fe3 0x100 0x11, 1
USB device 2FE3010014400000 - will be reset!
000397.390821 AppleUSBHostCompositeDevice@: AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe0005000)
USB device 2FE3010014400000 - failed to re-instantiate!
USB device 2FE3010014400000 - USB Mass Storage stack will be terminated!
USB device 2FE3010014400000 - fConsecutiveResetCount = 1.
MBP-di-Damiano:~$ sudo dmesg
```

PR #15646 (2019-05-12): usb: handle endpoint-sized EP0 IN transactions
Fixes #15507
closed 2019-05-12 by nashif