zephyrproject-rtos/zephyr #15513: nRF timer unnecessary configuration?

pabigot, 2019-04-17:
While working on my alarm infrastructure I noticed that nrf_timer_rtc for the system clock uses nrf_rtc_event_enable() for COMPARE0, which is the basic timer event, and has done so since the module was originally implemented in 2016.

As I understand it enabling events is intended to introduce a dependency on HFCLK to support use with PPI, but it increases power consumption when HFCLK is not otherwise required.  AFAICT nobody should ever connect that event to a PPI except possibly for whitebox instrumentation.

Is that configuration step appropriate?

nordic-krch, 2019-04-17:
Not sure if I understand the issue but enabling event is required to get interrupt from that COMPARE0 event. It has to be done even if PPI is not used.

pabigot, 2019-04-17:
@nordic-krch No.  Enabling INTEN is necessary to get the interrupt.  Enable EVTEN is necessary for only for PPI.

pabigot, 2019-04-20:
[Nordic devzone](https://devzone.nordicsemi.com/f/nordic-q-a/14023/evten-inten-nrf52-rtc) has a very clear description of the rolls of `EVTEN` versus `INTEN`.

The [nRF52840 Product Specification](https://infocenter.nordicsemi.com/topic/ps_nrf52840/rtc.html?cp=3_0_0_5_21_5#concept_eml_dkj_sr) explains why channels should be disabled in `EVTEN` if PPI is not being used.

nordic-krch, 2019-04-23:
I didn't know that then.

> The nRF52840 Product Specification explains why channels should be disabled in EVTEN if PPI is not being used.

Frankly speaking, to me this section doesn't mention PPI and drawing shows that `EVTEN` register gates PPI and IRQ so i'm confused. I guess you verified it so it's rather documentation issue.

pabigot, 2019-04-23:
Agree the diagram is a little confusing; the tie to PPI is in the text at the end of the arrow at the bottom.

I've been doing system timers with Nordic for over four years now and never had to touch the EVTEN registers, which is why I was a bit confused when I looked at what that HAL function did.  I've certainly verified it's not necessary.

PR #15624 (2019-04-25): drivers: timer: nrf: remove unnecessary event feature
Enabling the RTC event is intended to support peripheral-to-peripheral
interconnects, so introduces a request for HFCLK and PCLK16M when the
event is triggered.  This specific event is never used with PPI so
enabling events apparently does nothing but increase power consumption.

Closes #15513
commit 68a950a8 (2019-04-25): drivers: timer: nrf: remove unnecessary event feature

Enabling the RTC event is intended to support peripheral-to-peripheral
interconnects, so introduces a request for HFCLK and PCLK16M when the
event is triggered.  This specific event is never used with PPI so
enabling events apparently does nothing but increase power consumption.

Closes #15513
closed 2019-04-25 by galak