zephyrproject-rtos/zephyr #15516: Implementation of CONFIG_MAX_PTHREAD_COUNT

SteveOss, 2019-04-17:
**Describe the bug**

CONFIG_MAX_PTHREAD_COUNT appears to have been implemented as the maximum number of POSIX threads that can ever be created, rather than what might be expected i.e the maximum number of active POSIX threads. If the former is intended IMHO this is a "bad idea"
in that it breaks any dynamic creation of threads and should be documented, if the latter then is clearly a bug.

**To Reproduce**

In a loop call pthread_create on a function that immediately returns i.e. the thread starts then
terminates, after a few iterations the function returns EAGAIN as the maximum thread count
has been exceeded.

**Expected behavior**

I would expect to be able to create POSIX threads and only get an  EAGAIN error if I exceed
a configured maximum of *running* threads. The POSIX use of EAGAIN implies this i.e. try
again later and you may be able to create a thread.

**Impact**

Breaks standard TCP handling program, where listener creates a new thread to handle a connection.

**Environment (please complete the following information):**
 
Am using the v1.14-branch GA release branch and testing on a frdm-k64f, although looking at the source is not platform specific.

**Additional context**

In lib/posix/pthread.c take a look at use of pthread_num (as a minimum should be an atomic variable or accessed via a mutex, even for existing semantics).

Fix is straight forward:

- Create a mutex to protect pthread_num and associated stack array (posix_thread_pool)
- Add an array to mark usage of posix_thread_pool (e.g. boolean array)
- On thread creation, iterate usage array to find free pool element and increment pthread_num
- On thread exit decrement pthread_num and mark pool element as free.

Probably about 20 lines of code :)

wentongwu, 2019-04-18:
not related to your reported issue, did you enable CONFIG_USERSPACE when using POSIX threads?

SteveOss, 2019-04-18:
CONFIG_USERSPACE was not configured.

wentongwu, 2019-04-19:
https://github.com/wentongwu/zephyr/commit/8b6edc1c36fadcd9d9ac5dec042f860cda28ba8e
@SteveOss  would you like try this patch? Thanks!

SteveOss, 2019-04-19:
@wentongwu Patch looks good. We have a bank holiday here today and Monday, so won't be able
till test till Tuesday. Will let you know how I get on.

wentongwu, 2019-04-23:
@SteveOss  I test this patch on my side, it works well, how about your test?

SteveOss, 2019-04-23:
@wentongwu Have pulled your commit and is working for me. Guess the thread pool index could have been stored in the posix_thread struct to avoid the lookup on thread exit, but is no big deal. Thanks for sorting this, is this going to be back ported to the 1.14 release branch?

SteveOss, 2019-05-10:
Is this fix going back onto the 1.14 TLS branch ?

PR #15618 (2019-05-11): lib/posix: correct the meaning of CONFIG_MAX_PTHREAD_COUNT
CONFIG_MAX_PTHREAD_COUNT seems to be implemented as the maximum number
of POSIX threads that can ever be created, rather than the maximum
number of active POSIX threads. Create an array to mark usage of
posix_thread_pool. On thread creation, use this array to find free pool
element and on thread exit mark posix_thread_pool element as free.

Fixes #15516.
commit 6922d93d (2019-05-11): lib/posix: correct the meaning of CONFIG_MAX_PTHREAD_COUNT

Current code implement CONFIG_MAX_PTHREAD_COUNT as the maximum number
of POSIX threads that can ever be created, rather than the maximum
number of active POSIX threads. Use pthread_state of struct posix_thread
to track the state of posix thread in posix_thread_pool so that we can
reuse the unused posix thread.

Fixes #15516.
closed 2019-05-11 by nashif