zephyrproject-rtos/zephyr #15522: Extra padding in IPv4 link local ARP packets

pfl, 2019-04-17:
Link local IPv4 address configuration ARP packets have extra padding at the end. Packets captured by wireshark look like the following:
```
0000   ff ff ff ff ff ff fc c2 3d 12 f3 ab 08 00 00 01
0010   08 00 06 04 00 01 fc c2 3d 12 f3 ab 00 00 00 00
0020   00 00 00 00 00 00 a9 fe 19 00 00 00 00 00 00 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 
```
The whole issue is described in #15373

tbursztyka, 2019-04-17:
I guess in arp while preparing the arp message it adds up arp hdr size again to an already set up pkt (since ipv4 autoconf does it by itself).

pfl, 2019-04-17:
> I guess in arp while preparing the arp message it adds up arp hdr size again to an already set up pkt (since ipv4 autoconf does it by itself).

The third patch in #15433  was attempting to do that, but when I added a link local check to net_buf_add(pkt->buffer, sizeof(struct net_arp_hdr)) in arp_prepare() in ethernet/arp.c, no packet was sent out. So something

jukkar, 2019-04-25:
I tried to replicate this with latest master but did not see the issue. I tested it like this:
* `cd samples/net/ipv4_autoconf`
* `cmake -B build -DBOARD=qemu_x86 -DOVERLAY_CONFIG=../sockets/echo_server/overlay-e1000.conf .`
* `make -C build/ run`

Perhaps the extra padding is added by the ethernet driver. @pfl what board you were using in your test?

jukkar, 2019-04-25:
Export from wireshark
```
No.     Time           Source                Destination           Protocol Length Info
      5 6.903332147    RealtekU_12:34:56     Broadcast             ARP      42     Who has 169.254.64.76? Tell 0.0.0.0

Frame 5: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface 0
Ethernet II, Src: RealtekU_12:34:56 (52:54:00:12:34:56), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Address Resolution Protocol (request)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: request (1)
    Sender MAC address: RealtekU_12:34:56 (52:54:00:12:34:56)
    Sender IP address: 0.0.0.0
    Target MAC address: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Target IP address: 169.254.64.76

No.     Time           Source                Destination           Protocol Length Info
      6 8.913311421    RealtekU_12:34:56     Broadcast             ARP      42     Gratuitous ARP for 169.254.64.76 (Request)

Frame 6: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface 0
Ethernet II, Src: RealtekU_12:34:56 (52:54:00:12:34:56), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Address Resolution Protocol (request/gratuitous ARP)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: request (1)
    [Is gratuitous: True]
    Sender MAC address: RealtekU_12:34:56 (52:54:00:12:34:56)
    Sender IP address: 169.254.64.76
    Target MAC address: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Target IP address: 169.254.64.76
```

pfl, 2019-05-07:
This is a specific bug in SAM E70 Xplained. With this board I get
```
0000   ff ff ff ff ff ff fc c2 3d 12 f3 ab 08 06 00 01
0010   08 00 06 04 00 01 fc c2 3d 12 f3 ab 00 00 00 00
0020   00 00 00 00 00 00 a9 fe 8b 38 00 00 00 00 00 00
0030   00 00 00 00 00 00 00 00 00 00 00 00

```
This is a 60-byte ARP packet, when the correct Ethernet frame on the wire should be 42. Adding some logging reveals
```
[00:00:12.260,000] <inf> eth_sam: eth_tx: frag 0x204064ec len '14'
[00:00:12.260,000] <inf> eth_sam: eth_tx: frag 0x2040646c len '28'
[00:00:12.260,000] <inf> net_ethernet: ethernet_send: pkt len '42'
```
Thus the bug must be somewhere in the SAM E70 Xplained ethernet driver, not in the IP or ethernet stack. The log output was from a patch like this:
```
diff --git a/subsys/net/ip/ipv4_autoconf.c b/subsys/net/ip/ipv4_autoconf.c
index ba25f60cc1..b472167061 100644
--- a/subsys/net/ip/ipv4_autoconf.c
+++ b/subsys/net/ip/ipv4_autoconf.c
@@ -37,6 +37,10 @@ static struct net_pkt *ipv4_autoconf_prepare_arp(struct net_if *iface)
        /* We provide AF_UNSPEC to the allocator: this packet does not
         * need space for any IPv4 header.
         */
+
+       LOG_INF("%s: alloc buffer size '%zd'",
+               __FUNCTION__, sizeof(struct net_arp_hdr));
+
        pkt = net_pkt_alloc_with_buffer(iface, sizeof(struct net_arp_hdr),
                                        AF_UNSPEC, 0, BUF_ALLOC_TIMEOUT);
        if (!pkt) {
diff --git a/subsys/net/l2/ethernet/ethernet.c b/subsys/net/l2/ethernet/ethernet.c
index 7ef252de4e..fb4b633d14 100644
--- a/subsys/net/l2/ethernet/ethernet.c
+++ b/subsys/net/l2/ethernet/ethernet.c
@@ -609,6 +609,7 @@ send:
        ethernet_update_tx_stats(iface, pkt);
 #endif
        ret = net_pkt_get_len(pkt);
+       LOG_INF("%s: pkt len '%d'", __FUNCTION__, ret);
        ethernet_remove_l2_header(pkt);
 
        net_pkt_unref(pkt);
```

aurel32, 2019-05-07:
This just look like normal Ethernet padding for packets smaller than 64 bytes (header + user data + FCS). Note that Wireshark does not show the FCS, that's why the packets you observe are 60 bytes long.

pfl, 2019-05-08:
> This just look like normal Ethernet padding for packets smaller than 64 bytes (header + user data + FCS). Note that Wireshark does not show the FCS, that's why the packets you observe are 60 bytes long.

It is interesting that only this board produces said padding in Wireshark. All others inlcuding Linux produce 42 byte ARP packets, and so does Zephyr when it sends ARP messages elsewhere(!). So while it looks right, I'm not sure it ends up being right...

aurel32, 2019-05-08:
> It is interesting that only this board produces said padding in Wireshark. All others inlcuding Linux produce 42 byte ARP packets, and so does Zephyr when it sends ARP messages elsewhere(!). So while it looks right, I'm not sure it ends up being right...

Then that's a bug of those boards/drivers, not of the SAM E70 GMAC driver.

The Ethernet frame minimum size is 64 bytes, the packets must be padded otherwise they are considered as collisision. It's also the behavior I observe on Linux host when the received packets are seen as 60 bytes long. Of course the ARP packets not sent through Ethernet does not get padded or get padded differently. The same way the ARP packets sent by the host where wireshark is running are not padded, because they are tapped before they reach the media.

On the SAM E70 GMAC device, the padding is done by default in hardware, as explained in section 38.6.3.4 Transmit AHB Buffers. It can be disabled, by setting bit 16 of the transmit buffer descriptor entry, but then Zephyr has to do the padding and the CRC computation for the packet to be valid.

aurel32, 2019-05-08:
Just tested on an NXP IMXRT1020 board, I observe the same behavior, ie the ARP packages are padded to 60 bytes.