zephyrproject-rtos/zephyr #15565: undefined references to `sys_rand32_get'

lorepieri8, 2019-04-19:
**Describe the bug**
I get `undefined references to `sys_rand32_get'` after building a zephyr app with qemu_cortex_m3 or qemu_x86, while the same setup seems working for nrf52840_pca10056.

**To Reproduce**
Steps to reproduce the behavior:
1. https://github.com/AnyLedger/anyledger-wallet, commit 8fb0247e1ac3d61d20802eaf2f81c2511039174e
2. zephyr build instructions, with zephyr version picked using:
```
git clone https://github.com/zephyrproject-rtos/zephyr.git
DATE='Nov 7 2018'
git rev-list -1 --before="$DATE" master | xargs -Iz git checkout z
```
3. cmake -GNinja -DBOARD=qemu_cortex_m3 -DBUILD_XCOMPILE=1 ../
4. ninja
5. See error

**Expected behavior**
ninja completing the build

**Screenshots or console output**
```
libapp.a(compat.c.obj): In function `random32':
/Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/samples/anyledger/anyledger-wallet/build/../zephyr/compat.c:26: undefined reference to `sys_rand32_get'
zephyr/libzephyr.a(coap.c.obj): In function `coap_next_token':
/Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/subsys/net/lib/coap/coap.c:1484: undefined reference to `sys_rand32_get'
/Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/subsys/net/lib/coap/coap.c:1485: undefined reference to `sys_rand32_get'
zephyr/libzephyr.a(slip.c.obj): In function `slip_iface_init':
/Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/drivers/net/slip.c:488: undefined reference to `sys_rand32_get'
zephyr/subsys/net/ip/libsubsys__net__ip.a(net_context.c.obj): In function `find_available_port':
/Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/subsys/net/ip/net_context.c:98: undefined reference to `sys_rand32_get'
zephyr/subsys/net/ip/libsubsys__net__ip.a(net_if.c.obj):/Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/subsys/net/ip/net_if.c:2008: more undefined references to `sys_rand32_get' follow
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: MacOS
 - ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
 -GNUARMEMB_TOOLCHAIN_PATH="~/opt/gcc-arm-none-eabi-7-2018-q2-update/"

**Additional context**

It works with 
`cmake -GNinja -DBOARD= nrf52840_pca10056 -DBUILD_XCOMPILE=1 ../`
while it doesn't work for 
`cmake -GNinja -DBOARD= qemu_x86 -DBUILD_XCOMPILE=1 ../`

jukkar, 2019-04-19:
Try to set these in prj.conf file
```
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
```

lorepieri8, 2019-04-19:
Thanks @jukkar.

1- This indeed fixes the qemu_cortex_m3 ninja. How did you figure out about those config options? Where can I get more info on how to identify which config=y I need?

2- qemu_x86 is still failing. I think that the problem is the toolchain that I'm using. 

```[14/219] Building C object zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj 
ccache /Users/lorenzo/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../../kernel/include -I../../../../arch/x86/include -I../../../../soc/x86/ia32 -I../../../../soc/x86/ia32/include -I../../../../soc/x86/include -I../../../../boards/x86/qemu_x86 -I../../../../include -I../../../../include/drivers -Izephyr/include/generated -I../../../../lib/libc/minimal/include -I../../../../ext/lib/fnmatch/. -I../../../../subsys/net/lib/coap/. -I../../../../subsys/net/lib/lwm2m/. -isystem /Users/lorenzo/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include -isystem /Users/lorenzo/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/include-fixed -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/samples/anyledger/anyledger-wallet/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -std=c99 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj   -c /Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/arch/x86/core/offsets/offsets.c
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s: Assembler messages:
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:36: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:42: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:48: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:54: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:60: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:66: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:72: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:78: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:84: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:90: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:96: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:102: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:108: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:114: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:120: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:126: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:132: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:138: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:144: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:150: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:156: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:162: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:168: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:174: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:180: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:187: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:193: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:199: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:205: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:211: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:217: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:223: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:229: Error: unrecognized symbol type ""
/var/folders/f1/mytd0rsx76lbmj1fdgvwcp_40000gn/T//ccnprfN3.s:235: Error: unrecognized symbol type ""
[... cut ...]
[15/219] Performing download step (git clone) for 'trezor-crypto-lib'
Cloning into 'trezor-crypto'...
Already on 'master'
Your branch is up to date with 'origin/master'.
Submodule 'tests/wycheproof' (https://github.com/google/wycheproof) registered for path 'tests/wycheproof'
Cloning into '/Users/lorenzo/gitHubRepo/nov7zephyr/zephyr/samples/anyledger/anyledger-wallet/build/thirdparty/trezor-crypto/tests/wycheproof'...
Submodule path 'tests/wycheproof': checked out '2904be69e9d666bf3064fdc15093747e695cfae6'
ninja: build stopped: subcommand failed.
```

andrewboie, 2019-04-22:
>  I think that the problem is the toolchain that I'm using.

You can't compile qemu_x86 with a gcc-arm-none-eabi-7-2018-q2-update compiler.

jukkar, 2019-04-23:
> How did you figure out about those config options?

I just looked existing networking samples for clues, and these two options were used in many places when compiling for qemu.

lorepieri8, 2019-04-23:
I see. I hoped there was a simpler way. 
I will open a feature request. It would be useful to know which config options are needed for given functions.

closed 2019-04-23 by lorepieri8

dieselburner, 2020-05-08:
FYI: I get this error on current master when building google mqtt demo for `nucleo_f756zg` target.