zephyrproject-rtos/zephyr #15580: SAMD21 Adafruit examples no longer run on boards

ixmeins, 2019-04-20:
A fresh download and install of Zephyr 14 will build the (eg) samples/subsys/shell/shell_module code but when downloaded to an Adafruit Feather M0 board the code does not run or simply causes the soc to reboot. On Zephyr 13 these examples compiled and ran as expected.
The helloworld sample produces the same result, successful build but just reboots the board into bootloader on 14.
I found this after trying to re-add my SAMR21 soc and board to the new Zephyr build but could not get anything to run. It appears to be a new issues in the samd21 soc support.
Compiler is 8.2.1 20181213 (release) [gcc-8-branch revision 267074]

ixmeins, 2019-04-20:
Fixed - another newb gotcha, you need to manually set the flash base address (ninja menuconfig) to 0x2000 to allow for the Adafruit feather bootloader being at 0x0000. Obviously I did that once when I first installed 13.

closed 2019-04-20 by ixmeins

KubaFYI, 2020-01-15:
Are we expected to need to change this manually every single time? Is there a way to make that into a permanent change?