zephyrproject-rtos/zephyr #15587: Zephyr was unable to find the toolchain

lorepieri8, 2019-04-21:
**Describe the bug**
I'm not able to build sample projects such as "blinky", since Zephyr is unable to find the toolchain.

**To Reproduce**
Steps to reproduce the behavior:

export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR=/root
source zephyr-env.sh
Go to /zephyr/samples/basic/blinky
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=qemu_cortex_m3 ../
3. See error

**Impact**
Cannot build the project.

**Screenshots or console output**
```
root@3e5b46fa1a8b:/zephyrproject/zephyr/samples/basic/blinky/build# cmake -DBOARD=qemu_cortex_m3 ../
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Selected BOARD qemu_cortex_m3
Zephyr version: 1.13.99

warning: UART_CONSOLE (defined at drivers/console/Kconfig:36) was assigned the value 'y' but got the
value 'n'. You can check symbol information (including dependencies) in the 'menuconfig' interface
(see the Application Development Primer section of the manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE.html (which is updated
regularly from the master branch). See the 'Setting configuration values' section of the Board
Porting Guide as well.

warning: UART_STELLARIS (defined at drivers/serial/Kconfig.stellaris:1) was assigned the value 'y'
but got the value 'n'. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_STELLARIS.html
(which is updated regularly from the master branch). See the 'Setting configuration values' section
of the Board Porting Guide as well.
Parsing Kconfig tree in /zephyrproject/zephyr/Kconfig
Loading /zephyrproject/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3_defconfig as base
Merging /zephyrproject/zephyr/samples/basic/blinky/prj.conf
CMake Error at /zephyrproject/zephyr/cmake/compiler/gcc.cmake:18 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: zephyr

  Internal variables:

  CROSS_COMPILE:
  /root/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-

  TOOLCHAIN_HOME: /root/sysroots/x86_64-pokysdk-linux

Call Stack (most recent call first):
  /zephyrproject/zephyr/cmake/toolchain.cmake:71 (include)
  /zephyrproject/zephyr/cmake/app/boilerplate.cmake:272 (include)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04 (docker)
 - Toolchain: Zephyr SDK, installed in "/root"
 - Commit SHA or Version used: latest

**Additional context**
I've tried installing zephyr SDK also in "/" or in "/opt/zephyr-sdk/", but the result is the same.

Content of "/root":

```
root@3e5b46fa1a8b:~# cd /root
root@3e5b46fa1a8b:~# ls
arc-zephyr-elf   cmake-3.14.2-Linux-x86_64     info-zephyr-sdk-0.10.0    protobuf            sdk_version
arm-zephyr-eabi  cmake-3.14.2-Linux-x86_64.sh  nanopb                    protobuf-3.6.1      sysroots
cmake            i586-zephyr-elf               nanopb-0.3.9.3-linux-x86  protoc3             xtensa-zephyr-elf
cmake-3.14.2     i586-zephyr-elfiamcu          nios2-zephyr-elf          riscv32-zephyr-elf
```

carlescufi, 2019-04-21:
This looks like remnants of an old version of zephyr, because `/zephyrproject/zephyr/cmake/compiler/gcc.cmake` doesn't seem to exist anymore? can you run `git clean -fdx` from the root of your repo (careful, this will delete any files not tracked by Git) and `git rev-parse HEAD` to show us what commit you are in?

lorepieri8, 2019-04-22:
```
root@3e5b46fa1a8b:/zephyrproject/zephyr# git rev-parse HEAD
b7515ab0151250ba04514b57c1f2e7bf081cb504
root@3e5b46fa1a8b:/zephyrproject/zephyr# git show b7515ab0151250ba04514b57c1f2e7bf081cb504
commit b7515ab0151250ba04514b57c1f2e7bf081cb504 (HEAD)
Author: Himanshu Jha <himanshujha199640@gmail.com>
Date:   Thu Oct 25 15:18:03 2018 +0530
```

This was indeed the problem, I can build now. Thank you.

closed 2019-04-22 by lorepieri8