zephyrproject-rtos/zephyr #15588: Does zephyr support different time slices for each thread?

dwj-scut, 2019-04-21:
There are two threads of the same priority, one that takes longer time to execute and the other that takes shorter. Can we assign two different time slice to each other?

dwj-scut, 2019-04-21:
I didn't find the method in the document of zephyr

carlescufi, 2019-04-21:
@dwj-scut yes, you need to enable `CONFIG_TIMESLICING`: https://docs.zephyrproject.org/latest/reference/kernel/threads/index.html#preemptive-time-slicing

andrewboie, 2019-04-24:
This is not currently supported, there are no per-thread timeslice values.

andyross, 2019-04-24:
But that said, there's no particular reason it can't be if there's an actual need.  The slice time is set on context switch (or interrupt exit into a different thread) and we can fetch it from anywhere.

But I'd worry about a design that relied on this. Timeslicing is a (crude) way to achieve something approximating "fair" distribution of the global CPU resources.  It sounds from the phrasing above that you're trying to use it to manage real-time processes according to their expected run times.  That's not likely to work well.  Consider using some other trick like timeouts or workqueues if you want to manage individual work items like this.  Or just have your threads block with the appropriate frequency when their work is finished, and you won't need timeslicing to get fair distribution -- they'll take what they need.

ioannisg, 2019-07-30:
I think this question was answered. @ghost please, reopen if needed.

closed 2019-07-30 by ioannisg