zephyrproject-rtos/zephyr #15591: Add STM32 LCD-TFT Display Controller (LTDC) Driver

matthew-koch, 2019-04-22:
**Is your feature request related to a problem? Please describe.**
Add support for STM32's LTDC on STM32F4, F7, H7, etc.

**Describe the solution you'd like**
Initially, an API specific to the LTDC implemented with a long-term objective of enhancing the generic display API to support more advanced LCD drivers either on-board or in-MCU while supporting smaller displays with little overhead. This driver should be modular; able to be built such that unused feature sets can be removed. The STM32 HAL will not be used (greenfield implementation). Existing generic display API will have shortcomings (e.g., pixel format support). Use shim to provide API implementation; native and generic APIs will be exclusive from each other.

Would like to support:

- Compositing layers with transparency (LTDC supports 2 layers).
- Multiple color models (Monochrome, CLUT/indexed, RGB), color spaces and pixel formats.
- Specialized DMA (DMA2D) for fill/blit/blend operations, including pixel format conversion and alpha blending.
- Double buffering, able to utilize separate SRAM/SDRAM banks - configurable at buildtime or runtime. VSYNC timing/notification.
- MPU/D-Cache management for DMA operations
- Buildtime/runtime LCD configuration (buildtime at board config)
- Blanking/contrast/backlight control (need separate callbacks/drivers for this - power and backlight controls are on the board)

Other considerations for features not supported by LTDC, but are common (to support in generic API enhancements):
- Extensible color model/space/pixel format for packed (and planar?) YUV.
- Hardware accelerated stretch/scale.
- OS/API-enforced constraints, e.g. Layer zero dimensions same as display, layer reordering limitations, maximum supported display size, limited number of display driver instances,

**Describe alternatives you've considered**
Crayons attached to steppers.

**Additional context**

carlescufi, 2019-04-22:
> **Describe alternatives you've considered**
> Crayons attached to steppers.

🥇

matthew-koch, 2019-04-23:
FYI (no response required)

If it wasn't implied in the issue writeup, I intend to take a stab at this. I'm doing development on an stm32f746g_disco board.

Will be slow going to start - Zephyr's build system has a bit of a learning curve with some tools I'm not familiar with. However, nobody seems to be pounding the table for this driver. I will try to get a PR on a couple of things that might cause ugly merges later (pinmux, device tree) done first.

matthew-koch, 2019-05-04:
Quick update.

Made progress on the driver, but have gotten stuck by not having a working memory controller for external SDRAM. Didn't quite realize how sparse support for STM32 was before diving into this, so I'm having to switch gears and work on getting enough memory to handle the framebuffer and sort out the current memory mapping (DTCM vs. SRAM).

k0d, 2020-03-13:
@matthew-koch I'm working on this issue, are you still interested in having this working? I've got the memory side of things working, and have some code working for the LTDC which I'll commit in a day or 2 after cleaning it up.

matthew-koch, 2020-03-16:
By all means! I ran into a big project at work and haven't had a lot of
time for this. I kept chasing things that weren't implemented in Zephyr
(clocks, SDRAM, etc.) and this ballooned beyond what time I had. I won't be
able to pick this up again until late spring or early summer, so I'd
appreciate having this working.

- MK

DRNadler, 2020-06-12:
Any update on LTDC driver?
Thanks!
Best Regards, Dave

DRNadler, 2020-06-12:
@matthew-koch , @k0d - Any update on LTDC driver?
Thanks!
Best Regards, Dave

akshay-dandekar, 2020-09-28:
I am working on STM32F429I-DISC1. Above patches seems to be for STM32F7. Do they need to be modified for STM32F429?

Regards,
Akshay

tmilkovic51, 2022-01-19:
@matthew-koch , @k0d Are you still working on this issue? I would like to open a PR with my implementation if you aren't working on yours anymore.

I managed to get the driver working with LVGL sample on STM32H7B3I_DK board. The driver uses HAL library, and supports Kconfig-configurable pixel format and frame buffer location (internal/external SRAM). So far, it only supports blocking writes to frame buffer, but I plan to add non-blocking writes with DMA2D in the future.

k0d, 2022-01-19:
@tmilkovic51 I'm not working on it anymore...feel free to submit your code.

gmarull, 2022-08-11:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/display/display_stm32_ltdc.c

closed 2022-08-11 by gmarull