zephyrproject-rtos/zephyr #15593: How to use gdb to view the stack of a thread

xiaoliang314, 2019-04-22:
When I use gdb to interrupt the execution of the MCU, It usually stop in the idle thread. The stack of the idle thread is useless on me. 
At this time, I want to see what happened to my thread, what should I do? How should I use gdb to view the stack of my thread?

xiaoliang314, 2019-04-22:
I think I know how to do it.
This is my steps(ARM v7):
1. Find the thread(demo_thread) you need to view
2. Use the following command to switch the stack to this thread. 
```
gdb> set $lr = *(int*)(demo_thread->callee_saved.psp + 20)
gdb> set $pc = *(int*)(demo_thread->callee_saved.psp + 24)
gdb> set $psp = demo_thread->callee_saved.psp + 32
```
3. Use the 'bt' command to view the thread stack
```
gdb> bt
```

MaureenHelm, 2019-04-22:
Hi @xiaoliang314 , if you're using openocd or pyocd gdb server, you can enable zephyr thread awareness by setting `CONFIG_OPENOCD_SUPPORT=y` in your application. You can then use gdb thread commands.
https://docs.zephyrproject.org/latest/guides/debugging/host-tools.html
https://sourceware.org/gdb/current/onlinedocs/gdb/Threads.html

xiaoliang314, 2019-04-23:
@MaureenHelm Thanks for your reply, I am using J-Link GDB Server + GDB (ARM) for debugging. CONFIG_OPENOCD_SUPPORT has no effect on my debugging tools.

MaureenHelm, 2019-04-29:
The J-Link GDB server doesn't currently support zephyr thread awareness.

closed 2019-12-11 by MaureenHelm