zephyrproject-rtos/zephyr #15594: net: socket send failed when sent many times

chinli, 2019-04-22:
we used net shell of zephyr phylogenetic (version of 1.14.0-rc3-465)，and PC create TCP server, devices use command of "net tcp connect 192.168.1.107 8000" to connected, then successfully, send data by "net tcp send "hello world"" cmd, and repeate it  dozens of times. shell will report a error about failed to get free buffer. the detail information as below:

uart:~$ net tcp send "hello world "
##context_alloc_pkt##: pkt=0
Cannot send msg (-12)
[00:04:11.510,000] <dbg> net_buf.test_mem: (0x0010659c) 0x00120048	32	30	TX DATA (tx_bufs)

[00:04:11.510,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x0010659c) pkt_alloc_buffer():867: pool 0x00120048 size 256 timeout 1000
[00:04:11.570,000] <err> net_buf: net_pkt_get_reserve_data_debug():381: Failed to get free buffer
[00:04:11.570,000] <dbg> net_buf.test_mem: (0x0010b2b8) Fragment length 256 bytes

[00:04:11.570,000] <dbg> net_buf.test_mem: (0x0010b2b8) Network buffer pools:

[00:04:11.570,000] <dbg> net_buf.test_mem: (0x0010b2b8) Address		Total	Avail	Name

[00:04:11.570,000] <dbg> net_buf.test_mem: (0x0010b2b8) 0x0011fea4	16	16	RX

[00:04:11.570,000] <dbg> net_buf.test_mem: (0x0010b2b8) 0x0011fec0	16	14	TX

[00:04:11.570,000] <dbg> net_buf.test_mem: (0x0010b2b8) 0x0012001c	32	32	RX DATA (rx_bufs)

[00:04:11.570,000] <dbg> net_buf.test_mem: (0x0010b2b8) 0x00120048	32	30	TX DATA (tx_bufs)

[00:04:12.520,000] <err> net_buf: pkt_alloc_buffer():867: Failed to get free buffer
[00:04:12.520,000] <dbg> net_buf.test_mem: (0x0010659c) Fragment length 256 bytes

[00:04:12.520,000] <dbg> net_buf.test_mem: (0x0010659c) Network buffer pools:

[00:04:12.520,000] <dbg> net_buf.test_mem: (0x0010659c) Address		Total	Avail	Name

[00:04:12.520,000] <dbg> net_buf.test_mem: (0x0010659c) 0x0011fea4	16	16	RX

[00:04:12.520,000] <dbg> net_buf.test_mem: (0x0010659c) 0x0011fec0	16	14	TX

[00:04:12.520,000] <dbg> net_buf.test_mem: (0x0010659c) 0x0012001c	32	32	RX DATA (rx_bufs)

[00:04:12.520,000] <dbg> net_buf.test_mem: (0x0010659c) 0x00120048	32	30	TX DATA (tx_bufs)

[00:04:12.520,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x0010659c) pkt_alloc_buffer():867: pool 0x00120048 size 256 timeout 1000
[00:04:13.530,000] <err> net_buf: pkt_alloc_buffer():867: Failed to get free buffer
[00:04:13.530,000] <dbg> net_buf.test_mem: (0x0010659c) Fragment length 256 bytes

[00:04:13.530,000] <dbg> net_buf.test_mem: (0x0010659c) Network buffer pools:

[00:04:13.530,000] <dbg> net_buf.test_mem: (0x0010659c) Address		Total	Avail	Name

[00:04:13.530,000] <dbg> net_buf.test_mem: (0x0010659c) 0x0011fea4	16	16	RX

[00:04:13.530,000] <dbg> net_buf.test_mem: (0x0010659c) 0x0011fec0	16	14	TX

[00:04:13.530,000] <dbg> net_buf.test_mem: (0x0010659c) 0x0012001c	32	32	RX DATA (rx_bufs)

[00:04:13.530,000] <dbg> net_buf.test_mem: (0x0010659c) 0x00120048	32	30	TX DATA (tx_bufs)

[00:04:14.670,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x0010b2b8) net_pkt_get_reserve_data_debug():381: pool 0x00120048 size 256 timeout 100
Message sent
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) pkt_alloc_buffer():867: pool 0x0012001c size 256 timeout 100
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) allocated buf 0x0011aba0
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) pkt_alloc_buffer():867: pool 0x0012001c size 256 timeout 100
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) allocated buf 0x0011ab58
[00:04:14.720,000] <dbg> net_buf.net_buf_unref_debug: (0x0010b220) buf 0x0011aba0 ref 1 pool_id 2 frags 0x0011ab58
[00:04:14.720,000] <dbg> net_buf.net_buf_unref_debug: (0x0010b220) buf 0x0011ab58 ref 1 pool_id 2 frags 0x00000000
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) pkt_alloc_buffer():867: pool 0x0012001c size 256 timeout 100
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) allocated buf 0x0011ab58
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) pkt_alloc_buffer():867: pool 0x0012001c size 256 timeout 100
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) allocated buf 0x0011aba0
[00:04:14.720,000] <dbg> net_buf.net_buf_unref_debug: (0x0010b220) buf 0x0011ab58 ref 1 pool_id 2 frags 0x0011aba0
[00:04:14.720,000] <dbg> net_buf.net_buf_unref_debug: (0x0010b220) buf 0x0011aba0 ref 1 pool_id 2 frags 0x00000000
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) pkt_alloc_buffer():867: pool 0x0012001c size 256 timeout 100
[00:04:14.720,000] <dbg> net_buf.net_buf_alloc_len_debug: (0x00107248) allocated buf 0x0011aba0
[00:04:14.720,000] <dbg> net_buf.net_buf_unref_debug: (0x0010b220) buf 0x0011aba0 ref 1 pool_id 2 frags 0x00000000

pfalcon, 2019-04-22:
> (version of 1.14.0-rc3-465)

Please use 1.14.0 release, or master.

Beyond that, how reproducible is this? (I.e. of e.g. 20 runs of a board (which board?) after hardware reset, how many runs exhibit this issue?)

chinli, 2019-04-23:
@pfalcon tks for your caution! I upgrade the code, it's fixed! ^_^

closed 2019-04-23 by chinli