zephyrproject-rtos/zephyr #15598: Standard devicetree connectors for boards

Mani-Sadhasivam, 2019-04-22:
**Is your feature request related to a problem? Please describe.**
For the boards belonging to a standard form factor like 96Boards, it would be nice to have a dedicated devicetree connector which can expose the available peripherals like GPIO, I2C, SPI in a generic way so that the sample applications can run on any board adhering to the standard.

**Describe the solution you'd like**
This could be achieved by having a below devicetree node in board dts:

```
        96b-ls-con {
                compatible = "96b-ls-con";
                i2c0 = &i2c0;
                uart0 = &uart0;

                gpioa: gpioa {
                        gpios = <&gpio0 1 0>;
                        label = "GPIO-A";
                };
                gpiob: gpiob {
                        gpios = <&gpio0 2 0>;
                        label = "GPIO-B";
                };
        };
```

Then the applications can make use of the definitions like `DT_96B_LS_CON_GPIOA_GPIO_CONTROLLER`, `DT_96B_LS_CON_GPIOA_GPIO_PIN` etc... 

For achieving the above solution, a separate connector binding needs to be created and it shall include all possible peripherals.

**Describe alternatives you've considered**
There is already a concept of shields exist in Zephyr but it is limited to the expansion connectors and specifically not able to handle individual gpio's.

mike-scott, 2019-04-22:
There is a bit of discussion about supporting GPIOs for headers (and shields) on this example PR: https://github.com/zephyrproject-rtos/zephyr/pull/14057

erwango, 2019-04-24:
@Mani-Sadhasivam, I've created https://github.com/zephyrproject-rtos/zephyr/issues/15637 that aims at solving this issue

erwango, 2019-05-23:
@Mani-Sadhasivam, can you check https://github.com/zephyrproject-rtos/zephyr/pull/15861 and comment of solving your querry?

galak, 2019-09-06:
@Mani-Sadhasivam any update on this?

Mani-Sadhasivam, 2019-09-09:
Will check #15861 soon.

Mani-Sadhasivam, 2019-10-10:
@erwango I think this satisfies my requirement. This is what I tried:

```
/ {
        lscon_96b: connector {
                compatible = "96b-lscon-30pin";
                #gpio-cells = <2>;
                gpio-map = <13 0 &gpioa 8 0>,   /* GPIO-A */
                           <14 0 &gpiob 3 0>,   /* GPIO-B */
                           <16 0 &gpiob 5 0>,   /* GPIO-D */
                           <17 0 &gpioa 13 0>,  /* GPIO-E */
                           <18 0 &gpioa 14 0>,  /* GPIO-F */
                           <19 0 &gpioa 12 0>,  /* GPIO-G */
                           <20 0 &gpiob 4 0>;   /* GPIO-H */
        };
};

lscon_96b_i2c0: &i2c1 {};
lscon_96b_uart0: &usart1 {};
```
**NOTE** Devicetree spec doesn't allow the node label to start with a number, so I can't use `96b_lscon`.

And I have the following define in the `generated_dts_board.conf`
`DT_INST_0_96B_LSCON_30PIN=1`

Now, I'm just curious if you have any sample application for testing/validating this!

erwango, 2019-10-10:
> Now, I'm just curious if you have any sample application for testing/validating this!

@Mani-Sadhasivam, ASAFAIK no spevcific test, but indirectly new version of `gpio_basic_api` would help you to do that.

erwango, 2019-10-10:
@Mani-Sadhasivam, can we close the ticket then ?

Mani-Sadhasivam, 2019-10-10:
> @Mani-Sadhasivam, can we close the ticket then ?

@erwango I'll submit a PR for adding connector support to all 96Boards and will link this issue to that.

PR #19751 (2019-10-30): Add Low speed connector support for 96Boards IE boards
All 96Boards complying to the IE spec exposes either 40pin or 30pin
standard low speed connectors for peripheral connectivity. These
connectors are well defined and available in the IE spec [1]. This PR
creates two different gpio bindings for these connectors:

1. 96b-lscon-1v8 - 40 pin connector with maximum of 12 GPIOs
2. 96b-lscon-3v3 - 30 pin connector with maximum of 8 GPIOs

These bindings will be utilized by the 96Boards for exposing the
GPIO pins as nexus node as per the devicetree spec. This will allow
the shields and applications [... cut]
closed 2019-10-30 by galak