zephyrproject-rtos/zephyr #15601: pwm: nRF default prescalar value is wrong

doceme, 2019-04-22:
**Describe the bug**
A value of 1 for the ```data->prescalar``` causes the frequency to be half of desired frequency. I've confirmed this using an oscilloscope. The correct value for ```data->prescalar``` should be ```0U```. 

**To Reproduce**
1. Modify ```samples/basic/fade_led/src/main.c``` to set a PWM cycle value to something <= 32767. For example:
```
pwm_pin_set_cycles(pwm_dev, PWM_CHANNEL, 29088, 29088 / 2); // 550 Hz
```
2. build with board ```nrf52840_pca10056```
3. flash and observe actual frequency of LED is half of expected value

**Expected behavior**
A PWM value of desired frequency (not halved).

**Impact**
nRF PWM driver is not usable at certain frequencies.

**Screenshots or console output**
Oscilloscope screenshot with default ```data->prescalar``` set to ```1U```:

[image]

Oscilloscope screenshot with default ```data->prescalar``` set to ```0U```:

[image]

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: GNU Arm Embedded Toolchain (prebuilt version 7-2018-q2-update)
 - Commit SHA or Version used: v1.14.0

anangl, 2019-04-23:
@doceme Thanks for reporting this.
Obviously, when there is no need to divide the PWM clock, (i.e. the requested period cycles fit into the 15-bit PWM counter), the prescaler value should be `0`, not `1`. I'll provide a PR that fixes this.

doceme, 2019-04-23:
Thanks @anangl. I was going to make a PR, but got busy.

PR #15615 (2019-04-23): drivers: pwm_nrfx: Correct prescaler value for undivided clock
When there is no need to divide the PWM clock (i.e. the requested period cycles fit the 15-bit PWM counter), the prescaler value should be 0, not 1.

Fixes #15601.
PR #15629 (2019-06-11): [Backport v1.14-branch] drivers: pwm_nrfx: Correct prescaler value for undivided clock
Backport #15615.

Fixes #15601
closed 2019-04-23 by carlescufi