zephyrproject-rtos/zephyr #15603: Unable to use C++ Standard Library

mrfuchs, 2019-04-22:
**Describe the bug**
C++ Standard Library cannot be used with Zephyr 1.14.0. It used to work with 1.14-rc1.

**To Reproduce**
prj.conf: `CONFIG_CPLUSPLUS=y`

main.cpp:
```
#include <zephyr.h>
#include <vector>

int main()
{
  std::vector<int> v;

  v.push_back(10); 
  
  for (int x : v) 
    printk("x: %d", x); 

  return 0;
}
```

Command line: `D:\zephyr-test>cmake -GNinja -Bbuild -H. -DBOARD=stm32f4_disco && ninja -C build`

Output:
```
Zephyr version: 1.14.0
-- Found PythonInterp: C:/Users/user/AppData/Local/Programs/Python/Python37/
python.exe (found suitable version "3.7.0", minimum required is "3.4")
-- Selected BOARD stm32f4_disco
-- Loading D:/test/zephyr-v1.14.0/boards/arm/stm32f4_disco/stm32f4_disco.dts as
base
-- Overlaying D:/test/zephyr-v1.14.0/dts/common/common.dts
Parsing Kconfig tree in D:/test/zephyr-v1.14.0/Kconfig
Loading D:/test/zephyr-v1.14.0/boards/arm/stm32f4_disco/stm32f4_disco_defconfig
as base
Merging D:/zephyr-test/prj.conf
Configuration written to 'D:/zephyr-test/build/zephyr/.config'
-- Cache files will be written to: C:\Users\user\AppData\Local/.cache/zephyr

-- The C compiler identification is GNU 8.2.1
-- The CXX compiler identification is GNU 8.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Tools Arm Embedded/8 2018-q4-majo
r/bin/arm-none-eabi-gcc.exe
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: D:/zephyr-test/build
ninja: Entering directory `build'
...
../src/main.cpp:2:10: fatal error: vector: No such file or directory
 #include <vector>
          ^~~~~~~~
compilation terminated.
[13/121] Building C object zephyr/arch...__arm__core__cortex_m.dir/prep_c.c.obj
ninja: build stopped: subcommand failed.
```

**Expected behavior**
It compiles, as with 1.14-rc1:
```
D:\zephyr-test>cmake -GNinja -Bbuild -H. -DBOARD=stm32f4_disco && ninja -C build

Zephyr version: 1.14.0
-- Found PythonInterp: C:/Users/user/AppData/Local/Programs/Python/Python37/
python.exe (found suitable version "3.7.0", minimum required is "3.4")
-- Selected BOARD stm32f4_disco
-- Loading D:/test/zephyr-1.14.0-rc1/boards/arm/stm32f4_disco/stm32f4_disco.dts
as base
-- Overlaying D:/test/zephyr-1.14.0-rc1/dts/common/common.dts
Parsing Kconfig tree in D:/test/zephyr-1.14.0-rc1/Kconfig
Loading D:/test/zephyr-1.14.0-rc1/boards/arm/stm32f4_disco/stm32f4_disco_defconf
ig as base
Merging D:/zephyr-test/prj.conf
Configuration written to 'D:/zephyr-test/build/zephyr/.config'
-- Cache files will be written to: C:\Users\user\AppData\Local/.cache/zephyr

-- The C compiler identification is GNU 8.2.1
-- The CXX compiler identification is GNU 8.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Tools Arm Embedded/8 2018-q4-majo
r/bin/arm-none-eabi-gcc.exe
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
Including module(s):
-- Configuring done
-- Generating done
-- Build files have been written to: D:/zephyr-test/build
ninja: Entering directory `build'
[116/121] Linking CXX executable zephyr\zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       13592 B         1 MB      1.30%
            SRAM:        4348 B       192 KB      2.21%
        IDT_LIST:         120 B         2 KB      5.86%
[121/121] Linking CXX executable zephyr\zephyr.elf
```

**Impact**
Cannot update to Zephyr 1.14.0 for current development project.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: GNU ARM Embedded
 - Version used: Zephyr 1.14.0

mrfuchs, 2019-04-25:
Is this a reasonable fix or am I oversimplifying?

```diff
diff --git a/cmake/compiler/gcc/target_baremetal.cmake b/cmake/compiler/gcc/target_baremetal.cmake
index d268604435..0c657a61ac 100644
--- a/cmake/compiler/gcc/target_baremetal.cmake
+++ b/cmake/compiler/gcc/target_baremetal.cmake
@@ -5,7 +5,8 @@ macro(toolchain_cc_nostdinc)
 
   if (NOT CONFIG_NEWLIB_LIBC AND
     NOT COMPILER STREQUAL "xcc" AND
-    NOT CONFIG_NATIVE_APPLICATION)
+    NOT CONFIG_NATIVE_APPLICATION AND
+    NOT CONFIG_CPLUSPLUS)
     zephyr_compile_options( -nostdinc)
     zephyr_system_include_directories(${NOSTDINC})
   endif()
```

aescolar, 2019-04-26:
CC @SebastianBoe

carlescufi, 2019-05-14:
@SebastianBoe could you please take a look at this one?

SebastianBoe, 2019-05-15:
Presumably, this is due to NOSTDINC not pointing to the correct directory, but I am unable to figure out how to set it correctly, so avoiding -nostdinc is OK.

galak, 2019-05-15:
Take a look at the PR here and discussion, we might need to resurrect this: 

https://github.com/zephyrproject-rtos/zephyr/pull/13648

SebastianBoe, 2019-08-13:
The reporter has not enabled CONFIG_NEWLIB_LIBC.

To use the C++ standard library the CONFIG_LIB_CPLUSPLUS must be enabled. And to enable CONFIG_LIB_CPLUSPLUS, the CONFIG_NEWLIB_LIBC option must be enabled. A dependency that enforces this is now added in https://github.com/zephyrproject-rtos/zephyr/pull/18224

pabigot, 2019-08-14:
It is true that `CONFIG_NEWLIB_LIBC` is necessary, however, the problem description indicates that the `gnumarmemb` toolchain is also involved, so #17564 then triggers.  A backport of the patch to newlib from #18242 is required to fully solve the problem.

pabigot, 2019-09-30:
Backport is now in branch.

PR #18552 (2019-09-25): [Backport v1.14-branch] C++: fix builds with non-Zephyr toolchain
This backports the change and test case from #18242, which will close #15603 in the original poster's context of 1.14.
closed 2019-09-30 by pabigot