zephyrproject-rtos/zephyr #15604: Suspicious PCI and build_on_all default test coverage

marc-hb, 2019-04-22:
Like any other test suite, sanitycheck's default test coverage tries to strike a difficult balance between:
1. too little coverage, versus:
2. combinatorial explosion -> impractical and unusable build times.

This github issue is about two specific default coverage settings: A. **PCI** and B. **build_on_all**

Context: while performing some unrelated experimentations (don't ask) I broke by accident pci.c in my workspace. Then I ran sanitycheck which returned... 2 out 700 build failures: samples/synchronization/sample.synchronization on galileo and up_squared. I found the 2/700 number very surprising so I dug deeper and found this:

A. There is zero "default test platform" that compiles **PCI** code:
```
  git grep -B1 default:.*true -- boards/
```
Q: Should there be at least one platform?

B. sanitycheck supports a useful **build_on_all** test parameter. As just demonstrated above, `build_on_all` is useful to counter-balance the unavoidable limitation of the 'default test platforms" approach; 2 test failures is better than 0 test failures. It was probably meant for exactly that? However build_on_all is currently being used by... only one test. Or rather: only one toy "samples/" which feels less "serious" than something located in tests/ and is for sure dropped every time anyone passes any `-T tests/mysubsystem/` option to sanitycheck.

Q: Should there be a couple more and more "serious" tests with `build_on_all: on`? Don't get me wrong: build_on_all does increase build times and should be used very rarely but... just this toy "sample"?

marc-hb, 2019-04-22:
Found these more or less related issues:
Enhance Testing and Test Coverage #5027
Provide a mean to find tests with 0 platforms due to bad filtering #652
No test builds these files #12860
need to improve filtering and coverage in sanitycheck #6857

nashif, 2019-05-08:
> Q: Should there be at least one platform?

no, default platforms are primarily to do coverage on architecture level, not peripherals.

> Should there be a couple more and more "serious" tests with build_on_all: on? 

Maybe, any good suggestions?

marc-hb, 2019-05-09:
> no, default platforms are primarily to do coverage on architecture level, not peripherals.

I thought `pci.c` was not just a "mere peripheral" but somewhere between that and "architecture level".

> Maybe, any good [build_on_all] suggestions?

Not really; copying better informed people.

Wild guesses: tests/shell/, tests/ztest/base, tests/kernel/common/.

andrewboie, 2019-10-18:
> A. There is zero "default test platform" that compiles **PCI** code:
> 
> ```
>   git grep -B1 default:.*true -- boards/
> ```
> 
> Q: Should there be at least one platform?

I think so, yes.
Is this a situation where it's not turned on the board config, instead relying on tests and so forth to turn it on instead? Seems like it ought to be in the platform defconfig for Apollo Lake and Upsquared.

nashif, 2019-11-05:
we did enable tests on all plaforms in some other tests (kernel/), so this should be covered.

closed 2019-11-05 by nashif

marc-hb, 2019-11-06:
Yes, build_all was added to tests/kernel/common/testcase.yaml by commit fc4cb492b51

I purposely broke drivers/pcie/pcie.c and ran sanitycheck on HEAD = 9ff64bb49717, ignoring unrelated (C++?) failures these 6 tests failed:

```
 up_squared                samples/boards/up_squared/gpio_counter/sample.board.up_squared.gpio_counter 
 gpmrb                     samples/synchronization/sample.synchronization
 up_squared                samples/synchronization/sample.synchronization   
 up_squared                tests/kernel/common/kernel.common 
 gpmrb                     tests/kernel/common/kernel.common 
 up_squared                tests/arch/x86/info/arch.x86.info  
```