zephyrproject-rtos/zephyr #15607: nRF52: 2.4 GHz proprietary RF support

KwonTae-young, 2019-04-23:
I am trying to make a wireless keyboard using the `nRF52840`.

At first I tried to use another MCU + [nRF24L01](https://www.nordicsemi.com/?sc_itemid=%7B36BDC7E1-5E35-4967-8B4D-E41DCD96512F%7D).
However, the product description says `Not recommended for new designs`.
I do not know exactly why it is `Not recommended for new designs`.
[nRF24 Series](https://www.nordicsemi.com/?sc_itemid=%7B36BDC7E1-5E35-4967-8B4D-E41DCD96512F%7D)
[image]

So I'm looking at the `nRF52840`.
It is said that [nRF52 Series](https://www.nordicsemi.com/Products/Low-power-short-range-wireless/nRF52840#infotabs) supports `2.4 GHz proprietary`.  
[image]

Does Zephyr support the `2.4 GHz proprietary` driver?
I could not find a related source.

I'm sorry I do not have enough English. :)
Thank you.

carlescufi, 2019-04-23:
Hi @KwonTae-young. This feature is already supported in the zephyr-based SDK maintained by Nordic, take a look here: https://github.com/NordicPlayground/fw-nrfconnect-nrf/tree/master/samples/esb

closed 2019-04-23 by carlescufi

KwonTae-young, 2019-04-24:
@carlescufi
Thank you.
I did not know there was a separate SDK. :sweat_smile: