zephyrproject-rtos/zephyr #15608: [question] my board won't boot without debugger attached but no issue using nordic SDK

summer20100514, 2019-04-23:
I came across such weird situation which is my custom board couldn't boot without debugger. But the NRF52DK works well. I have tried to choose the internal RC clock, but in vain. Anyone who knows the root cause?
and what's more, if I use nordic mesh sdk, everything goes well without any problem. thank you.
@carlescufi @jhedberg @SebastianBoe @nvlsianpu

SebastianBoe, 2019-04-23:
I don't think there is enough information here for anyone to be able to give you a good answer. You are going to have to debug and find out what is going on.

summer20100514, 2019-04-23:
@SebastianBoe but if I attached the debugger, everything is ok. If I detached, the board seems not booting at all.

carlescufi, 2019-04-23:
@summer20100514 this sounds like it might be a clock issue perhaps? The debugger keeps the clock on I believe.

summer20100514, 2019-04-26:
@carlescufi seems due to DCDC settings.

closed 2019-08-05 by summer20100514