zephyrproject-rtos/zephyr #15612: bt_set_id_addr() to allow public address as argument

dwildmark, 2019-04-23:
**Is your enhancement proposal related to a problem? Please describe.**
As of the current state of Zephyr, one can't use bt_set_id_addr() to set a public address, as the function prohibits this. I see no reason why the API would prohibit this behavior. I would like to discuss why it is that way and if we come to the conclusion that it should be allowed, I will patch it.

**Describe the solution you'd like**
I want the bt_set_id_addr() function to allow the setting of public addresses.

**Describe alternatives you've considered**
I have not found any alternative solution.

I think @jhedberg might have some input on this matter, correct me if I'm wrong.

jhedberg, 2019-04-23:
Is the expectation that this would use the Zephyr vendor HCI extensions? There's no standard way to set the public address over HCI.

dwildmark, 2019-04-23:
Ok, now I see the problem. Was not aware that this is not possible through standard HCI commands.
The way I did it was a bit of a hack, but I basically pretended it was a Random address and disabled some checks in hci_core.c and it worked on the nRF52840, but it would probably not work on other bluetooth devices.
The vendor specific HCI extensions seems like the proper place for this kind of feature.

jhedberg, 2019-04-23:
For the record, the specification for the Zephyr HCI vendor command is available here: https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/reference/bluetooth/hci.txt#L408

carlescufi, 2019-04-23:
@jhedberg indeed, I was going to point that out. But the `bt_set_id_addr()` API call still needs to implement it right?

jhedberg, 2019-04-23:
> @jhedberg indeed, I was going to point that out. But the bt_set_id_addr() API call still needs to implement it right?

@carlescufi right

dwildmark, 2019-05-06:
Guys, closing this since I found out that bt_ctlr_set_public_addr() exists. At least for my use case, that is enough.

closed 2019-05-06 by dwildmark