zephyrproject-rtos/zephyr #15625: CMAKE_C_COMPILER_FORCED in boilerplate.cmake prevents usage of target_compile_features in CMakeLists.txt

gatzka, 2019-04-23:
I've a simple application that uses a library. In the libraries CMakeLists.txt I've the following statement:
`target_compile_features(sha PRIVATE c_std_99)`

When I now run west build, I got the following error during cmake configuration:
`target_compile_features no known features for C compiler
  "GNU"
  version 8.3.0`

After digging around, I found that this seems to be triggered by the statements
`set(CMAKE_C_COMPILER_FORCED   1)
set(CMAKE_CXX_COMPILER_FORCED 1)
`
in boilerplate.cmake.

If I set both variables to `0`, I can build my stuff. Unfortunately only for the native_posix target, which uses the host compiler. Using the toolchains from the zephyr SDK fails while trying to compile a simple test program.

I'm using the zephyr SDK for compilation, develop on Ubuntu 18.04, use west version 0.5.8  and zephyr master.

Is there a way out of this?

SebastianBoe, 2019-04-24:
Using 'target_compile_features' is not supported unfortunately.

c99 is already what is used by default in Zephyr, so I believe you can safely remove

target_compile_features(sha PRIVATE c_std_99)

gatzka, 2019-04-25:
> c99 is already what is used by default in Zephyr, so I believe you can safely remove
> target_compile_features(sha PRIVATE c_std_99)

Well, not really. The library I wrote is avaliable for several operating systems like Linux, Windows, mbedOS etc. For those it is highly beneficial to declare the C standard to be used.

SebastianBoe, 2019-04-25:
I'm not sure what kind of scenario you are describing, is it just the .a file that
is portable, or also the build scripts? If it's the build scripts then perhaps you
could write something like

if(NOT ZEPHYR)
target_compile_features(sha PRIVATE c_std_99)
endif()

ioannisg, 2019-07-30:
@gatzka please, re-open, if needed, to continue discussion

closed 2019-07-30 by ioannisg