zephyrproject-rtos/zephyr #15626: Link config options in prj.conf with enabled functions.

lorepieri8, 2019-04-23:
**Is your enhancement proposal related to a problem? Please describe.**
When building with `cmake -GNinja ` it can happen to have an error of the kind `In function X:
undefined reference to Y`
This problem can be solved enabling the right config options in prj.conf, but it's unclear a priori which option should be turned on. 

**Describe the solution you'd like**
An easy way to search with config options are needed to turn on function Y.

**Describe alternatives you've considered**
An error log mentioning: "have you considered turning on config option Z in order to enable Y?"

**Additional context**
The background of my request comes from this issue:
https://github.com/zephyrproject-rtos/zephyr/issues/15565

(feel free to give a better title to the issue)

SebastianBoe, 2019-04-24:
This would be good to have, ideas for how to do this in a maintainable way are welcome.

gordonklaus, 2019-07-23:
I have a couple of observations with regard to this issue.  Likely naïve (as I am very new to this project), but perhaps a good starting point:

- If you include a header file, then all function declared in it should also be defined.  That is, configuration should affect declarations and definitions equally.  An "undefined reference" error is a bug.  The absence of a function should be signaled by it not even being declared.
- If a certain configuration is to exclude some functions, then including the header file that declares those functions should immediately (i.e., via an `#error` directive) produce an error describing the reason for the conflict, i.e., the aforementioned "have you considered turning on config option Z in order to enable Y?" message.

I've no idea how maintainable this solution is.