zephyrproject-rtos/zephyr #15657: properly define kernel <--> arch APIs

andrewboie, 2019-04-24:
We are part of the way there, but we need a very clear definition of the interface between the core kernel, and the arch/ code.

- All APIs and macros prefixed with z_arch_ / Z_ARCH_
- All interface definitions in a single interface header under kernel/include/
- Pulling in arch/cpu.h has everything namespaced with either Z_ARCH/z_arch or z_<name of arch>

ioannisg, 2019-07-31:
I would very much like to get this into 2.0 @andrewboie; please, give a status update on this :)

andrewboie, 2019-07-31:
@ioannisg unsure if I'll get to this for 2.0. I haven't started it.
My highest priority remaining work for 2.0 is #14206 and #14683 
Let me see what I can do though, this one might not take too long.

dleach02, 2019-12-10:
Should this still be open?

@andrewboie

andrewboie, 2019-12-10:
Yup, I fixed this.

closed 2019-12-10 by andrewboie