zephyrproject-rtos/zephyr #15659: Add capability to generate an SBOM (based a files actually used) for each build.

kestewart, 2019-04-24:
**Is your feature request related to a problem? Please describe.**
Licensing of a binary is dependent on the licenses of the files included in the binary.   By generating an accurate SBOM based on the files actually included,  the licensing clarity is improved, and the compliance with the licensing is straight forward. 

**Describe the solution you'd like**
Generate an SPDX document as the output of each build, that summarizes all the source files included in the image.   If multiple images are generated (ie. from west),  then individual SPDX documents for each component are generated,  and an overview SPDX document generated to show relationship between the specific image SPDX documents. 

Suggest this be implemented as a specific "option" on cmake,  that will invoke the correct support to log the files included and use that as a basis for manifest for generation.   The licensing information can be extracted from each file by searching for "SDPX-License-Identifier" tags.

**Describe alternatives you've considered**
Creating an SPDX document will let the information be transformed and consumed by other tools.   Other formats as output are tool specific.

SebastianBoe, 2019-04-25:
It might be possible to extract this from CMake target properties, but it won't be easy.

It might also be possible to do "Build Output Parsing" and extract this from the logs of a verbose ninja build. But this probably won't be 100% safe.

One tricky use-case would be a source file that does

#include <../../../far_out_of_the_repo/into/new/repo/wrong_license.h>

point being that whatever mechanism one chooses, it would probably be best-effort, and not security-level safe.

endian-benjamin, 2019-05-03:
Set EXPORT_COMPILE_COMMANDS on. Or use 'bear' tool. Or strace.

henrikbrixandersen, 2019-10-20:
@kestewart Good idea. However, the suggested way to of doing this (extracting the SPDX tag from each compiled source file) will not work well for 3rd party modules, where we cannot enforce an SPDX tag in each source file. I have no good solution for that.

mbolivar-nordic, 2021-05-26:
@kestewart can we close this now that `west spdx` is merged?